STMicroelectronics/STM32CubeH7 #1: Missing source codes in this repository

engycz, 2019-05-20:
Repository not as described 

> STM32Cube MCU Full Package for the STM32H7 series - (HAL + LL Drivers, CMSIS Core, CMSIS Device, MW libraries plus a set of Projects running on all boards provided by ST (Nucleo, Evaluation and Discovery Kits))

Missing source codes.

ASELSTM, 2019-05-20:
Dear Jiří Engelthaler,

Thank you for your interest. We are presently working on this. We aim to have all STM32Cube packages available on Github soon.

We cannot share a date for the moment. We will keep you informed.

With regards

ALABSTM, 2019-08-05:
Hi engycz,

STM32CubeH7 package (v1.5.0) has been published a couple of weeks ago. We hope this addresses your request. In such case this issue could be closed.

With regards,

closed 2019-08-05 by ALABSTM