STMicroelectronics/STM32CubeH7 #2: HRTIM_TIM_TypeDef is undefined

salkinium, 2019-06-04:
**Describe the set-up**
 * arm-none-eabi-gcc
 * `#include <stm32h7xx.h>`

**Describe the bug**

`HRTIM_TIM_TypeDef` is undefined. After cross referencing this with the RM you probably meant to use `HRTIM_Timerx_TypeDef`.

**How To Reproduce**

Trying to access `HRTIM1_TIMA` in any code will fail, since `HRTIM_TIM_TypeDef` is undefined:

```c
#include <stm32h7xx.h>
HRTIM1_TIMA->TIMxCR = 0;
```

**Additional context**

```diff
-#define HRTIM1_TIMA         ((HRTIM_TIM_TypeDef *) HRTIM1_TIMA_BASE)
-#define HRTIM1_TIMB         ((HRTIM_TIM_TypeDef *) HRTIM1_TIMB_BASE)
-#define HRTIM1_TIMC         ((HRTIM_TIM_TypeDef *) HRTIM1_TIMC_BASE)
-#define HRTIM1_TIMD         ((HRTIM_TIM_TypeDef *) HRTIM1_TIMD_BASE)
-#define HRTIM1_TIME         ((HRTIM_TIM_TypeDef *) HRTIM1_TIME_BASE)
+#define HRTIM1_TIMA         ((HRTIM_Timerx_TypeDef *) HRTIM1_TIMA_BASE)
+#define HRTIM1_TIMB         ((HRTIM_Timerx_TypeDef *) HRTIM1_TIMB_BASE)
+#define HRTIM1_TIMC         ((HRTIM_Timerx_TypeDef *) HRTIM1_TIMC_BASE)
+#define HRTIM1_TIMD         ((HRTIM_Timerx_TypeDef *) HRTIM1_TIMD_BASE)
+#define HRTIM1_TIME         ((HRTIM_Timerx_TypeDef *) HRTIM1_TIME_BASE)
```

[Complete patch](https://github.com/modm-io/cmsis-header-stm32/blob/master/patches/h7.patch)

CCASTM, 2019-06-05:
Hello Niklas,
Thank you for your finding. This has been communicated to the development team and we will come back to you (a very first analysis confirms the problem)
Regards
Christophe

ALABSTM, 2019-06-19:
Hi Niklas,

The issue you pointed out has been confirmed. A fix will be available in a future release. This closes the issue.

With regards,

ALABSTM, 2019-06-19:
ST Internal Reference: 67769

closed 2019-06-19 by ALABSTM