STMicroelectronics/STM32CubeH7 #3: SDMMC_BASE is undefined

salkinium, 2019-06-04:
**Describe the set-up**
 * arm-none-eabi-gcc
 * `#include <stm32h7xx.h>`

**Describe the bug**

This line is wrong, since there is no `SDMMC` peripheral on the device.
```c
#define SDMMC               ((SDMMC_TypeDef *) SDMMC_BASE)
```

The correct peripherals are called `SDMMC1` and `SDMMC2` and they are correctly defined.

**How To Reproduce**

Try to call `SDMMC` in your code, which will fail since `SDMMC_BASE` is undefined.

**Additional context**

```diff
-#define SDMMC               ((SDMMC_TypeDef *) SDMMC_BASE)
```
[Full patch](https://github.com/modm-io/cmsis-header-stm32/blob/master/patches/h7.patch)

Also note that there is a use of a *signed* 32-bit integer in the `ETH_TypeDef` struct (around line 762):
```diff
   uint32_t      RESERVED15[14];
   __IO uint32_t MMCTSCGPR;
   __IO uint32_t MMCTMCGPR;
-  int32_t       RESERVED16[5];
+  uint32_t      RESERVED16[5];
   __IO uint32_t MMCTPCGR;
   uint32_t      RESERVED17[10];
   __IO uint32_t MMCRCRCEPR;
```

CCASTM, 2019-06-05:
Hello Niklas,
Thank you for your finding. This has been communicated to the development team and we will come back to you to confirm or explain this point.
Regards
Christophe

ASELSTM, 2019-06-18:
ST Internal Reference: 68309

ASELSTM, 2019-06-18:
Hi Niklas,

Thank you once again for your contribution. The problem was identified and confirmed, the correction will be available in our next release.

With regards

closed 2019-06-18 by ASELSTM

CCASTM, 2019-06-18:
To be more accurate, the following line will be removed (as not relevant)
#define SDMMC               ((SDMMC_TypeDef *) SDMMC_BASE)

But there is a second question in this Issue regarding ETH_TypeDef.
**Niklas**, can you create a new Issue, as it is more convenient to separate the corrections (better to track the commits)

Thank you
Christophe

salkinium, 2019-06-18:
You can also create issues 😛  \*delegates back\*