STMicroelectronics/STM32CubeH7 #5: TCP/IP sample for STM32H743ZI_Nucleo is unstable

illishar, 2019-09-30:
The TCP/IP sample code for STM32H743ZI_Nucleo doesn't work, as intended: (STM32CubeH7 v. 1.5) 
It's best described here:
https://community.st.com/s/question/0D50X00009ZExEkSAL/stm32h743-nucleo-ethernet-stops-working

ALABSTM, 2019-09-30:
Hi Illishar,

The issue you pointed out has been communicated to our development teams. It will be investigated soon. We will be back to you as soon as we get and answer.

With regards,

ALABSTM, 2019-10-08:
Hi Illishar,

Would you mind providing us with a simple test case other than the one you described on ST Community allowing to reproduce the issue?

Thank you,

ALABSTM, 2019-10-08:
Hi Illishar,

Could you please confirm you are using the latest firmware version (STM32CubeH7 v1.5.0) and that the issue is systematically reproducible on your side?

Thank you,

Sprutten, 2019-10-09:
Hi ALABSTM

I am the junior developer mentioned in this tread. Just to clarify, we are using the STM32H743ZI2_Nucleo boards and not STM32H743ZI_Nucleo. The latest firmware version (STM32CubeH7 v1.5.0) is installed. I am using this example for reference STM32Cube_FW_H7_V1.5.0\Projects\NUCLEO-H743ZI\Applications\LwIP\LwIP_HTTP_Server_Netconn_RTOS 

What I did to reproduce the issue:
Run the program and make a request to the server, then wait about 5-10 seconds and make a new request. Then the browser is loading forever and can’t get the HTML from the server. The problem is consistent after waiting 5-10 seconds. The example works fine if you make a request every 1-2 seconds.

What I have tested:
•	Use multiple STM32H743ZI2_Nucleo boards 
•	When testing the sample code on different PC setups, the issue is inconsistent. (More than one PC have this issue)
•	Use different browsers. Browsers differs in stability, but Chrome is more unstable.
•	Tried with static and DHCP option, issue occurs on both
•	Use CycloneTCP example. I can’t reproduce the error in this TCP stack. 
•	Use different network adapters and cables (USB, PCI, switch/no switch and built-in)

I have some Wireshark logs to confirm the issue. I have added some comments on the logs from the ST example, see below. 

In line 7 we can see the GET / HTTP/1.1 request isn’t answered by the server. I did a new request at line 12 about 5 seconds after the first request and that got answered. Then in line 69 I did a new request after about 10 seconds, to make sure it was reproduceable and that didn’t get answered by the server. 

I have attached the Wireshark logs from the ST example and CycloneTCP example. (CycloneTCP is included for reference)
[Wireshark logs.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/3706304/Wireshark.logs.zip)

ALABSTM, 2019-10-10:
Hi Sprutten,

Thank you for all these details. We will communicate them to our developments team in charge of investigating this issue. We may eventually ask you for other details if needed. Thank you in advance for your patience and help.

With regards,

ALABSTM, 2019-10-18:
Hi Sprutten,

We tried to reproduce the issue you described using the same example and the same board as you. Unfortunately we could not. Investigation is ongoing.

Just to be sure we understood the sequence you described, by "doing a new request", do you mean reloading the Web page on the browser by pressing F5?

Thank you for your answer and in case something new has been spotted from your side, that could shed more light about this behavior, do not hesitate to inform us. Thank you once more.

With regards,

Sprutten, 2019-10-18:
Hi ALISTM
It was when making request to the board. The same result occur if I press f5 or if I press the homepage in the html.

ALABSTM, 2019-10-18:
Hi Sprutten,

May I ask for more precision about how you make requests to the board? Thank you for your answer and for your patience. The point is that maybe our development team is missing some detail that prevents them from reproducing the issue.

With regards,

Sprutten, 2019-10-18:
Hi ALISTM
I make request to the board when loading the HTML page in a browser. 
The when the code is uploaded to the board, i open a browser and type the IP address the board has. I tried to make the same HTTP request with a C# script, it worked perfect and I could't reproduce the error. It only occurs when making requests in a browser.

I have discovered that if i make a delay(5ms) or a printf at this line in the hal ethernet driver it works. but thats not feasible since the the system becomes slow.
https://github.com/STMicroelectronics/STM32CubeH7/blob/e261d820b398846a94f22f4aeb32d86c29546efb/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_eth.c#L951

It also works if i make the same delay at this place.
https://github.com/STMicroelectronics/STM32CubeH7/blob/e261d820b398846a94f22f4aeb32d86c29546efb/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c#L282

I think the error occurs when modifying the transmit descriptor (TDES3 normal descriptor (Read format)). There might be some problem with the cache, but not sure.

Hope it helps.

ALABSTM, 2019-11-06:
Hi Sprutten,

We tried once more to reproduce the issue you have described, with no success. We used Google Chrome Version 76.0.3809.100 (Official Build) (32-bit) as browser.

We sent requests to the MCU by pressing F5, by clicking on the "Reload this page" button of the browser, and by clicking on the "Home page" link on the Web page. We did this each 7 seconds, with no success, however.

A suggestion from our development teams is that you increase the timeout value to fit more your application needs:
https://github.com/STMicroelectronics/STM32CubeH7/blob/e261d820b398846a94f22f4aeb32d86c29546efb/Projects/NUCLEO-H743ZI/Applications/LwIP/LwIP_HTTP_Server_Netconn_RTOS/Src/ethernetif.c#L45

Please let me know whether it solves the problem.

With regards,

Sprutten, 2019-11-06:
Hi ALISTM
Increasing the DMA timeout doesn't fix the issue. Still have the same issue.

Sprutten, 2019-11-06:
Hi ALISTM
Increasing the DMA timeout doesn't fix the issue. My guess to a solution, is to add a synchronization barrier inside the HAL driver after filling in the TxDesc and before restarting the DMA engine from the suspended state. Think there is some sort of sync issues. I have tried to add a DSB in the transmit frame without any luck. https://github.com/STMicroelectronics/STM32CubeH7/blob/e261d820b398846a94f22f4aeb32d86c29546efb/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_eth.c#L951
I think you have to test it on multiple PC's, since the issue is occurring on specific PC's and can't be replicated by PC in which the browser is working.

ALABSTM, 2019-11-06:
Hi Sprutten,

Would you mind sending the binary (or hex) file you are using to flash your MCU?

Thank you,

Sprutten, 2019-11-06:
Sure, here it is. 

[STM32H743ZI_Nucleo(bin).zip](https://github.com/STMicroelectronics/STM32CubeH7/files/3814170/STM32H743ZI_Nucleo.bin.zip)

ALABSTM, 2019-11-06:
Hi Sprutten,

We will also need the *.map file corresponding to the binary file you have sent.

Thank you,

Sprutten, 2019-11-06:
Sure, here it is. I also included other debug files.
[Debug.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/3815791/Debug.zip)
With regards

ALABSTM, 2019-11-08:
Hi Sprutten,

Thank you for the artifacts you have sent. Unfortunately, they were of no great help, since you are using a fixed IP address, our development teams say.
We could not perform the test as the network addresses are not the same. It would have been possible if the LWIP_DHCP option were activated.

There is one last thing we could do, however. Would you, please, send us your application code? Just the application code please. We will use it to perform the test again.

With regards,

Sprutten, 2019-11-08:
Ohh, sorry about the DHCP option. But it's literally the stock firmware code example i am using (STM32CubeH7 v1.5.0). The only thing i have changed is the DHCP option.
But here it is.
[LwIP_HTTP_Server_Netconn_RTOS.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/3824136/LwIP_HTTP_Server_Netconn_RTOS.zip)
If you want here is the updated binary file where the DCHP is enabled.
[Debug.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/3824151/Debug.zip)

The issue is also occurring when using stock code.

ALABSTM, 2019-11-15:
Hi Sprutten,

I hope you are doing well. Any update about the issue you are facing?

From our side, we could not reproduce the problem, even though we remade the test several times. According to our development teams, our middleware is cleared and the root cause of this issue comes most likely from elsewhere.

As the support we provide cannot go beyond the scope of the firmware we published on GitHub, please allow me to close this issue.

We are sorry not to be able to be of more help. We wish you good luck to solve this problem you are facing.

With regards,

Sprutten, 2019-11-15:
Hi Ali
First of all, thanks for taking your time to try and help me. Unfortunately the issue still persist and I guess i'll look into other examples/solutions.
It took me 8 different computers to reproduce the error on another machine but, it is what it is. 
Have a nice weekend.

With regards

ALABSTM, 2019-11-15:
Hi Sprutten,

I close the ticket for the moment. I really hope you could find the root cause of this issue and, ideally, solve it.

Anyway, if you ever become able to reproduce systematically the faulty behavior and find out that the problem is related to our middleware, do not hesitate to recontact me. We could reopen this issue again.

Sorry once more for not being able to help you more than we did and wish you the best of lucks.

With regards,

closed 2019-11-15 by ALABSTM