STMicroelectronics/STM32CubeH7 #6: Broken I2S extension module when using the 1.5.0 version.

liarokapisv, 2019-11-07:
It appears that the current `stm32h7xx_hal_i2s_ex.c` file is broken due to undeclared identifiers and CMSIS register usage. This can be verified by doing a recursive search for the identifiers in the repository.

These are the errors that occur when simply including the `stm32h7xx_hal_i2s_ex.h` file after defining `SPI_I2S_FULLDUPLEX_SUPPORT`:

```
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:243:17: error: 'HAL_I2S_STATE_BUSY_TX_RX' undeclared (first use in this function); did you mean 'HAL_SPI_STATE_BUSY_TX_RX'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:250:21: error: 'SPI_TypeDef' {aka 'struct <anonymous>'} has no member named 'DR'; did you mean 'SR'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:74: error: 'I2S2ext_BASE' undeclared (first use in this function); did you mean 'I2C2_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:105: error: 'I2S3ext_BASE' undeclared (first use in this function); did you mean 'I2C3_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:52:86: error: 'SPI_I2SCFGR_I2SE' undeclared (first use in this function); did you mean 'SPI_I2SCFGR_I2SDIV'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:280:25: error: 'SPI_TypeDef' {aka 'struct <anonymous>'} has no member named 'DR'; did you mean 'SR'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:377:40: error: 'SPI_TypeDef' {aka 'struct <anonymous>'} has no member named 'DR'; did you mean 'SR'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:463:21: error: 'HAL_I2S_STATE_BUSY_TX_RX' undeclared (first use in this function); did you mean 'HAL_SPI_STATE_BUSY_TX_RX'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:74: error: 'I2S2ext_BASE' undeclared (first use in this function); did you mean 'I2C2_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:105: error: 'I2S3ext_BASE' undeclared (first use in this function); did you mean 'I2C3_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:475:21: error: 'SPI_TypeDef' {aka 'struct <anonymous>'} has no member named 'DR'; did you mean 'SR'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:52:86: error: 'SPI_I2SCFGR_I2SE' undeclared (first use in this function); did you mean 'SPI_I2SCFGR_I2SDIV'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:574:21: error: 'HAL_I2S_STATE_BUSY_TX_RX' undeclared (first use in this function); did you mean 'HAL_SPI_STATE_BUSY_TX_RX'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:74: error: 'I2S2ext_BASE' undeclared (first use in this function); did you mean 'I2C2_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:105: error: 'I2S3ext_BASE' undeclared (first use in this function); did you mean 'I2C3_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:603:43: error: 'SPI_CR2_RXDMAEN' undeclared (first use in this function); did you mean 'SPI_CFG1_RXDMAEN'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:607:81: error: 'SPI_TypeDef' {aka 'struct <anonymous>'} has no member named 'DR'; did you mean 'SR'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:610:34: error: 'SPI_CR2_TXDMAEN' undeclared (first use in this function); did you mean 'SPI_CFG1_TXDMAEN'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:613:36: error: 'SPI_I2SCFGR_I2SE' undeclared (first use in this function); did you mean 'SPI_I2SCFGR_I2SDIV'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:640:63: error: 'SPI_TypeDef' {aka 'struct <anonymous>'} has no member named 'DR'; did you mean 'SR'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:74: error: 'I2S2ext_BASE' undeclared (first use in this function); did you mean 'I2C2_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:105: error: 'I2S3ext_BASE' undeclared (first use in this function); did you mean 'I2C3_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:74: error: 'I2S2ext_BASE' undeclared (first use in this function); did you mean 'I2C2_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:105: error: 'I2S3ext_BASE' undeclared (first use in this function); did you mean 'I2C3_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:885:49: error: 'SPI_CR2_RXDMAEN' undeclared (first use in this function); did you mean 'SPI_CFG1_RXDMAEN'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:913:40: error: 'SPI_CR2_TXDMAEN' undeclared (first use in this function); did you mean 'SPI_CFG1_TXDMAEN'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:947:35: error: 'SPI_CR2_RXDMAEN' undeclared (first use in this function); did you mean 'SPI_CFG1_RXDMAEN'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:947:53: error: 'SPI_CR2_TXDMAEN' undeclared (first use in this function); did you mean 'SPI_CFG1_TXDMAEN'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:74: error: 'I2S2ext_BASE' undeclared (first use in this function); did you mean 'I2C2_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:105: error: 'I2S3ext_BASE' undeclared (first use in this function); did you mean 'I2C3_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:973:19: error: 'SPI_TypeDef' {aka 'struct <anonymous>'} has no member named 'DR'; did you mean 'SR'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:74: error: 'I2S2ext_BASE' undeclared (first use in this function); did you mean 'I2C2_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:105: error: 'I2S3ext_BASE' undeclared (first use in this function); did you mean 'I2C3_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:1031:43: error: 'SPI_TypeDef' {aka 'struct <anonymous>'} has no member named 'DR'; did you mean 'SR'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:74: error: 'I2S2ext_BASE' undeclared (first use in this function); did you mean 'I2C2_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:105: error: 'I2S3ext_BASE' undeclared (first use in this function); did you mean 'I2C3_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:74: error: 'I2S2ext_BASE' undeclared (first use in this function); did you mean 'I2C2_BASE'?
../STM32Cube/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:46:105: error: 'I2S3ext_BASE' undeclared (first use in this function); did you mean 'I2C3_BASE'?
```

Here is a simplified list of the errors after removing duplicates and paths:
```
'HAL_I2S_STATE_BUSY_TX_RX' undeclared
'SPI_TypeDef' has no member named 'DR'
'I2S2ext_BASE' undeclared
'I2S3ext_BASE' undeclared
'SPI_I2SCFGR_I2SE' undeclared
'SPI_CR2_RXDMAEN' undeclared
'SPI_CR2_TXDMAEN' undeclared
```

Here are the results of running relevant rgreps in the repository's root:
```
rgrep HAL_I2S_STATE_BUSY_TX_RX
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:  hi2s->State = HAL_I2S_STATE_BUSY_TX_RX;
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:  hi2s->State     = HAL_I2S_STATE_BUSY_TX_RX;
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:  hi2s->State     = HAL_I2S_STATE_BUSY_TX_RX;

rgrep I2S2ext_BASE
Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:#define I2SxEXT(__INSTANCE__) ((__INSTANCE__) == (SPI2)? (SPI_TypeDef *)(I2S2ext_BASE): (SPI_TypeDef *)(I2S3ext_BASE))

rgrep I2S3ext_BASE
Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:#define I2SxEXT(__INSTANCE__) ((__INSTANCE__) == (SPI2)? (SPI_TypeDef *)(I2S2ext_BASE): (SPI_TypeDef *)(I2S3ext_BASE))

rgrep SPI_I2SCFGR_I2SE
[... 1 lines cut ...]
Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_i2s_ex.h:#define __HAL_I2SEXT_DISABLE(__HANDLE__) (I2SxEXT((__HANDLE__)->Instance)->I2SCFGR &= ~SPI_I2SCFGR_I2SE)
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:    if ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SE) != SPI_I2SCFGR_I2SE)
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:    if ((hi2s->Instance->I2SCFGR & SPI_I2SCFGR_I2SE) != SPI_I2SCFGR_I2SE)

rgrep SPI_CR2_RXDMAEN
[... cut ...]
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:  CLEAR_BIT(hi2s->Instance->CR2, (SPI_CR2_RXDMAEN | SPI_CR2_TXDMAEN));
Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_i2s_ex.c:  CLEAR_BIT(I2SxEXT(hi2s->Instance)->CR2, (SPI_CR2_RXDMAEN | SPI_CR2_TXDMAEN));

```

As you can see there are indeed no definitions of these variables.
There is also the issue of `'SPI_TypeDef' has no member named 'DR'`. I assume this is a leftover
from some other driver version. There is no such member in the `SPI_TypeDef`

ALABSTM, 2019-11-11:
Hi Liarokapisv,

Thank you for your contribution. Do you confirm you have this issue using the latest published firmware version (i.e. v1.5.0)? Thank you for your answer.

With regards,

liarokapisv, 2019-11-12:
Confirmed with both the CubeMX provided v1.5.0 package and this repository's master branch.

ALABSTM, 2019-11-12:
Hi Liarokapisv,

Thank you for the confirmation. I will forward this issue to our development teams. I will be back to you as soon as I get an answer.

With regards,

ALABSTM, 2019-11-13:
Hi Liarokapisv,

Here is what the reply of our development teams:
- The I2S full duplex was mentioned as a missed feature in the release notes of the v1.4.0 (***Known Limitations*** section).
- The I2S full duplex feature is not available in the v1.5.0 neither. The files stm32h7xx_hal_i2s_ex.c/.h are protected with a compilation define `#if defined (SPI_I2S_FULLDUPLEX_SUPPORT)`.
- This `SPI_I2S_FULLDUPLEX_SUPPORT` define is **no set by default**. This is to avoid compiling the full duplex feature implemented in these files but not yet ready to be used.

You have probably activated it (and you were right as we provided this feature within the driver). Hence,  you got these errors. We do apologize for this "ambiguity".

This issue will be fixed in two steps:
1. In the next release (v1.6.0), we will remove the stm32h7xx_hal_i2s_ex.c/.h from the package.
2. Once the full duplex correctly implemented, we will put these files back into the package to make this feature available to use.

Thank you once more for having tackled this point.

With regards,

ALABSTM, 2019-11-14:
ST Internal Reference: 63001

closed 2020-08-03 by ALABSTM