STMicroelectronics/STM32CubeH7 #8: enabling unwanted global interrupts

ashchpepkov, 2019-11-21:
In file stm32h7xx_hal_lptim.c function void LPTIM_Disable(LPTIM_HandleTypeDef *hlptim) use __disable_irq() and __enable_irq() without checking PRIMASK state.

ALABSTM, 2019-11-21:
Hi Ashchepkov,

Thank you for your remark. It will be forwarded to our development team to check its relevance. You will be updated as soon as the provide us with feedback. Thank you once more.

With regards,

ALABSTM, 2019-11-26:
ST Internal Reference: 76518

ALABSTM, 2019-11-26:
Hi Ashchepkov,

Our development teams confirmed the issue you pointed out. A fix will be implemented and made available in the frame of a future release.

We cannot share a date for the moment, but hopefully it should be soon. Thank you for your patience and thank you once more for your contribution.

With regards,

ALABSTM, 2020-04-15:
Hi @ashchepkov,

I hope you are doing well. I simply want to share a **sketch** of the fix proposed by development teams. Please note also that the same issue has been identified at the level of the `LL_LPTIM_Disable()` function.

Unfortunately, the fix is still not planned to be released in the next couple of weeks. We count on your patience and comprehension. Take care and stay safe.

With regards,

```diff
void LPTIM_Disable(LPTIM_HandleTypeDef *lptim)
{
  ...
+  uint32_t prim;

+  /* Read PRIMASK register, check interrupts status before disabling them */
+  prim = __get_PRIMASK();

  /* Disable interrupts */
  __disable_irq();

  ...

+  /* In case interrupts were enabled, set them back */
+  if (prim == 0)
+  {
    __enable_irq();
+  }
}
```

ashchepkov, 2020-04-16:
Hello!
I use the same method.

With best regards.

ALABSTM, 2021-12-14:
Hi @ashchepkov,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.9.1** of the STM32CubeH7 published recently on GitHub. Thank you again for having reported.

With regards,

closed 2021-12-14 by ALABSTM