STMicroelectronics/STM32CubeH7 #9: FreeRTOS cannot enable timer

cloud019, 2019-11-28:
Hi ST,

I try to enable freeRTOS timer by set the configUSE_TIMERS = 1 in the FreeRTOSConfig.h, but it can't work. The same method in STM32L031, it works normal.
Did I miss something?

Regards,
Wei

ALABSTM, 2019-11-29:
Hi Cloud019,

Welcome to our GitHub organization. Please allow me to introduce @rahmanih who will answer your question.

With regards,

rahmanih, 2019-11-29:
Hi,

> I try to enable freeRTOS timer by set the configUSE_TIMERS = 1 in the FreeRTOSConfig.h, but it can't work.

Could you please give more details about "can't work"?
- Are you using the CMSIS-OS API or the FreeRTOS native one?
- Are you able to create the timer?
- are getting a crash, hang ?

regards
Haithem.

cloud019, 2019-11-29:
> Hi,
> 
> > I try to enable freeRTOS timer by set the configUSE_TIMERS = 1 in the FreeRTOSConfig.h, but it can't work.
> 
> Could you please give more details about "can't work"?
> 
> * Are you using the CMSIS-OS API or the FreeRTOS native one?
FreeRTOS native
> * Are you able to create the timer?
Below is my code to create timer
	/* Create Timer */
	osTimerDef(LEDFlashTimer, driverLEDFlashTimerCallback);
	driverLEDFlashTimerHandle = osTimerCreate(osTimer(LEDFlashTimer), osTimerPeriodic, NULL);
	/* Start Timer */
	osTimerStart(driverLEDFlashTimerHandle, 200);
> * are getting a crash, hang ?
I set a breakpoint in the "driverLEDFlashTimerCallback" callback function, but the breakpoint does not active.
> 
> regards
> Haithem.

rahmanih, 2019-12-02:
> I set a breakpoint in the "driverLEDFlashTimerCallback" callback function, but the breakpoint does not active.

is the **_driverLEDFlashTimerHandle_**   **NULL** ?, if it is the case then you should set a breakpoint in  the function "osTimerCreate()" and check why the timer is not correctly created.

cloud019, 2019-12-02:
> is the **_driverLEDFlashTimerHandle_** **NULL** ?, if it is the case then you should set a breakpoint in the function "osTimerCreate()" and check why the timer is not correctly created.

Yes, the handle is NULL.
But I trace the function

	TimerHandle_t xTimerCreate(	const char * const pcTimerName,			/*lint !e971 Unqualified char types are allowed for strings and single characters only. */
								const TickType_t xTimerPeriodInTicks,
								const UBaseType_t uxAutoReload,
								void * const pvTimerID,
								TimerCallbackFunction_t pxCallbackFunction )
	{
	Timer_t *pxNewTimer;

		pxNewTimer = ( Timer_t * ) pvPortMalloc( sizeof( Timer_t ) );

		if( pxNewTimer != NULL )
		{
			prvInitialiseNewTimer( pcTimerName, xTimerPeriodInTicks, uxAutoReload, pvTimerID, pxCallbackFunction, pxNewTimer );

			#if( configSUPPORT_STATIC_ALLOCATION == 1 )
			{
				/* Timers can be created statically or dynamically, so note this
				timer was created dynamically in case the timer is later
				deleted. */
				pxNewTimer->ucStaticallyAllocated = pdFALSE;
			}
			#endif /* configSUPPORT_STATIC_ALLOCATION */
		}

		return pxNewTimer;
	}

I add a breakpoint at the "prvInitialiseNewTimer", it can stop on it, so the pxNewTimer  should not be NULL, but the **_driverLEDFlashTimerHandle_** is NULL, I don't know why.

rahmanih, 2019-12-03:
>I add a breakpoint at the "prvInitialiseNewTimer", it can stop on it, so the pxNewTimer should not be NULL, but the driverLEDFlashTimerHandle is NULL, I don't know why.

I suspect that the `pxNewTimer = ( Timer_t * ) pvPortMalloc( sizeof( Timer_t ) );` call returns '0' but the compiler is not considering the '0' as "NULL"

Could you check that the `pxNewTimer` is equal to `0` after the `pvPortMalloc()` call?
which IDE are you using?

regards
Haithem.

cloud019, 2019-12-08:
Hi Haithem,

1) pxNewTimer is not NULL, it has an address value.
2) Return to main code, driverLEDFlashTimerHandle become 0

Regards,
Wei

ALABSTM, 2020-04-14:
Hi @cloud019,

Any update regarding your question? Could you be able to enable the timer?

With regards,

ALABSTM, 2020-06-05:
Hi @cloud19 and @h

ALABSTM, 2020-07-22:
Hi @cloud019 and @rahmanih,

Please allow me to close this thread as no activity has been registered since a couple of months. Thank you for your comprehension.

With regards,

closed 2020-07-22 by ALABSTM