STMicroelectronics/STM32CubeH7 #10: In BSP driver lan8742: endless loop in case of timeout

pavel-a, 2019-12-14:
Hello there

After this line:
https://github.com/STMicroelectronics/STM32CubeH7/blob/e261d820b398846a94f22f4aeb32d86c29546efb/Drivers/BSP/Components/lan8742/lan8742.c#L170

break is needed. Else, in case of PHY reset timeout, endless loop occurs.
As reported in the forum [here](https://community.st.com/s/question/0D50X0000BmleyqSQA/stm32h747idisco-with-lwip-not-working).

-- pa

ALABSTM, 2019-12-16:
Hi Pavel,

Glad to hear from you again. On the forum, I could not find the firmware version you are using and whether your code is generated by Cube MX or not.

Actually, we provided a template to follow suggesting the different pieces of information that could be needed by our development teams for their investigations. Would you mind restating this issue according to this template? (essentially for the software- and hardware-related details and other aspects that may not have been precised on the forum).

Thank you for your comprehension.

With regards,

pavel-a, 2019-12-16:
Ali, this was reported by other person on the forum and I've asked him to join here. 
Below please find description per your template. Best regards,  P.

**Describe the set-up**
 * The board: STM32H747I-DISCO
 * IDE or compiler: unknown. The project was built with the Cube library. 

**Describe the bug**
When LAN8742 Ethernet PHY reset operation times out for any reason, the LAN8742 driver loops instead of returning with proper status `LAN8742_STATUS_RESET_TIMEOUT`.

**How To Reproduce**
1. Indicate the global behavior of your application project

The program hangs.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)

lan8742.c

https://github.com/STMicroelectronics/STM32CubeH7/blob/e261d820b398846a94f22f4aeb32d86c29546efb/Drivers/BSP/Components/lan8742/lan8742.c#L158-L172

3. The use case that generates the problem

LAN8742 Ethernet PHY reset operation times out. Maybe because of wrong pinout, or electrical problem.

4. How we can reproduce the problem

Do visual review of the code.

--------------

ericsonj, 2019-12-16:
Hi 

I reported this issue in ST Community. I'm using OpenSTM32 IDE (Ac6) and example code from STM32H7 repository version V1.5.0. I used the application Projects/STM32H743I-EVAL/Applications/LwIP/LwIP_UDPTCP_Echo_Server_Netconn_RTOS as reference.

ALABSTM, 2020-01-02:
Hi Pavel, hi Ericson,

Thank you for these details. You message will be forwarded to our development teams to be analyzed. We will be back to you as soon as we get an answer.

With regards,

ALABSTM, 2020-01-06:
Hi guys,

Our development teams said this issue has been already fixed internally. The fix will be available withing revision v1.6.0 of the STM32CubeH7 firmware package, which, hopefully, will be release soon.

Thank you for your patience and thank you once again for your contribution.

With regards,

ALABSTM, 2020-02-03:
Hi Pavel, hi Ericson,

 STM32CubeH7 firmware package v1.6.0 has been published since a couple of weeks. Would you mind check whether the BSP driver within this new package revision solves the issue? If it is the case, this issue can be closed.

With regards,

ericsonj, 2020-02-11:
Hi ALABSTM, 

The problem was solved. I used the firmware package v1.5.0 and apply the suggestions in user manual UM2411 revision 4.

Thank you very much for your attention and time.

ALABSTM, 2020-02-11:
Hi EricsonJ,

Glad to know the issue has been solved. Please allow me to close this thread and thank you again for having reported the point.

With regards,

closed 2020-02-11 by ALABSTM

ALABSTM, 2020-02-11:
Hi Ericson,

Just to be sure. You said you have used "firmware package v1.5.0". Did you mean v1.6.0? Thank you for your answer.

With regards,

ericsonj, 2020-02-17:
Hi Ali LABBENE

I used the firmware package v1.5.0 but now I upgraded to v1.6.0 and the
ethernet works fine.

Thanks you.