STMicroelectronics/STM32CubeH7 #12: USB MSC Stack: Can't enumerate when disk is not ready

canhkha, 2019-12-20:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * The board (either ST RPN reference or your custom board): STM32H747I-EVAL
 * IDE or at least the compiler and its version: Iar 8.30

**Describe the bug**
Windows can't enumerate device when SD card is not ready

**How To Reproduce**
1. Indicate the global behavior of your application project

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)
USB Device Stack, MSC class

3. The use case that generates the problem
SD Card is not insert
4. How we can reproduce the problem
* Remove sd card from sd slot
* Run STM32Cube_FW_H7_V1.5.0\Projects\STM32H747I-EVAL\Applications\USB_Device\MSC_Standalone\EWARM\USBD_MSC.eww
* Plug USB to Windows PC

**Additional context**
If you have a first analysis or patch correction, thank you to share your proposal.
* The returning value in `int8_t SCSI_ProcessCmd(USBD_HandleTypeDef *pdev, uint8_t lun, uint8_t *cmd)` is not handled
* Ref #11 for the fix.

ALABSTM, 2020-01-02:
Hi Canhkha,

Thank you for your message. Please allow me to introduce @bouattay from our development teams. He will analyze the issue you described and the relevance of the solution you suggested.

Please note I took the initiative to remove a couple of lines from your original description as these lines are only intended to serve as a guideline for users to state their issues. Removing them increases the readability of your message.

With regards,

ALABSTM, 2020-01-06:
Hi Canhkha,

As already said by @bouattay, "Many changes have been added in our latest device lib v2.6.0 including a similar fix to what you have shared".

Hopefully, a new revision of the STM32CubeH7 will be published soon which will include the fix. Thank you for your patience and for your contribution once again.

With regards,

ALABSTM, 2020-01-06:
ST Internal Reference: 42353

closed 2020-08-03 by ALABSTM