STMicroelectronics/STM32CubeH7 #13: DSI/emWin crash on dual-core H7

m0nka, 2020-01-06:
Hello,

example affected:
/Projects/STM32H747I-DISCO/Applications/STemWin/STemWin_HelloWorld/CM7/STemWin/App/BASIC_HelloWorld.c

running on non-modified STM32H747I-DISCO board.

Without edit to this file, 'Hello world!' string is displayed on the screen, but there is very limited amount of time that the function can print to the LCD, after that i suspect a crash of the emWin or the DSI LCD driver.

So this code displays only one line:

void MainTask(void)
{
 int i;

  GUI_Clear();
  GUI_SetFont(&GUI_Font32_1);
  GUI_DispStringAt("Hello world!", (LCD_GetXSize()-100)/2, (LCD_GetYSize()-20)/2);

  for(i = 0; i < 0x5FFFFF; i++)
    	__asm(".hword 0x46C0");

  GUI_DispStringAt("Hello world!", (LCD_GetXSize()-100)/2, (LCD_GetYSize()-20)/2 + 40);

  while(1)
  {
  }
}

decreasing the delay to 0x5FFFF allows for the second line to be printed. So anything printed before that timeout is reached will be displayed, but nothing after that. The four LEDs that get toggled by the timer will still blink.

I have similar issue with other example projects (the Menu Launcher, the STemWin Demonstration)
that use similar version of the same DSI driver. 

My best guess would be:

- some type of watchdog running by default ?
- the M4 crashes maybe and indirectly affects the LCD print ?
- some unknown emWin library crash (i have used it on various ST and NXP chips and never seen this issue before).

Thank you! 
K.

m0nka, 2020-01-11:
Further to this i have followed the STemWin544.pdf, page 54 and added this instead of my code:

`void MainTask(void) {
int i = 0;
GUI_Init();
GUI_DispString("Hello world!");
while(1) {
GUI_DispDecAt( i++, 20,20,4);
if (i > 9999) {
i = 0;
}
}
}`

It would only count to 0468 and then stop refresh the screen. But the loop is still running, for example if i add BSP_LED_Toggle(LED4) inside;

ASELSTM, 2020-01-15:
Hi Krassi,

This issue has been forwarded to our development teams. We will be back to you as soon as they provide us with feedback. Thank you for your contribution.

With regards,

m0nka, 2020-01-18:
Thank you! 

So far i have been able to make only one of the examples to work for me (Menu Launcher), but with keeping it using the emWin Window Manager exclusively and monitoring the two interrupts toggle the onboard LEDs.

I see that LTDC interrupt is enabled in most of the emWin examples, but doesn't fire, only the DSI and DMA2D. So i keep the two LEDs blinking. When any of the above mentioned Refresh blockage happens, the LEDs will stop blink (i have also scope connected there, as it is a bit faster for the eye).

m0nka, 2020-02-13:
Hey guys,

You can close the issue. Not sure if bug or feature, but apparently in 5.44 you can not use the 2D drawing calls directly.

WM_SetCallback(WM_HBKWIN, _cbBk); needs to be created and in WM_PAINT event only the normal 2D calls could be issued (while WM_InvalidateWindow(WM_HBKWIN); in the main task loop).

Thanks for everyone who spent time on this.

BR.

closed 2020-02-13 by m0nka