STMicroelectronics/STM32CubeH7 #14: Regression in QSPI vs. v1.3.0

thomask77, 2020-01-14:
Hi!

**Describe the set-up**

We are using an STM32H743XIH6 (Rev. V) @ 480 MHz, on a custom board with a W25Q128JVEIQ flash in Quad-SPI mode. Our Toolchain is gcc-arm-none-eabi-9-2019-q4-major.

**Describe the bug**

We're using the HAL_QSPI_Transmit_DMA function.

When using CubeMX libraries v1.3.0, we could run QSPI with 120 MHz clock (Clock Prescaler 1).

After an upgrade to CubeMX libraries v1.5.0, we had to set the prescaler to 8, or even 16 to achieve stable operation, otherwise, HAL_QSPI_Transmit_DMA would fail. 

We can send you the CubeMX file if required.

**Additional context**

I haven't looked into the bug in detail yet, but I could narrow it down to this single change:

v1.5.0, bad:

https://github.com/STMicroelectronics/STM32CubeH7/blob/e261d820b398846a94f22f4aeb32d86c29546efb/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_qspi.c#L1333-L1334

v1.3.0, good:
```c
      __HAL_QSPI_ENABLE_IT(hqspi, QSPI_IT_TE);
```

After reverting the change, we can run with 120 MHz again.

What was the reason for the change in 1.5.0?
Perhaps you can already try to reproduce on an evaluation board.

ASELSTM, 2020-01-15:
Hi Thomask77,

Thank you for reporting this issue. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get an answer. Thank you in advance for your patience.

With regards,

ASELSTM, 2020-01-16:
Hi Thomask77,

Our development teams confirmed the issue you pointed out. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-01-16:
ST Internal Reference: 79394

closed 2020-02-13 by ASELSTM