STMicroelectronics/STM32CubeH7 #15: 'register' storage class specifier is incompatible with C++17

thomask77, 2020-01-16:
Hi!

The `register` storage class specifier was deprecated in C++11 and removed in C++17.

There are a few `register` keywords in some inline functions in `stm32h7xx_ll_adc.h`:

https://github.com/STMicroelectronics/STM32CubeH7/blob/e261d820b398846a94f22f4aeb32d86c29546efb/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h#L2264

So, if I include the `stm32h7xx_ll_adc.h` header file from C++, gcc will complain:

```
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h: In function 'uint32_t LL_ADC_DMA_GetRegAddr(ADC_TypeDef*, uint32_t)':                                                                                                                                                                                                                                 
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h:2264:21: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]                                                                                                                                                                                                             
 2264 |   register uint32_t data_reg_addr;                                                                                                                                                                                                                                                                                                                                                  
      |                     ^~~~~~~~~~~~~                                                                                                                                                                                                                                                                                                                                                   
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h: In function 'void LL_ADC_SetCalibrationLinearFactor(ADC_TypeDef*, uint32_t, uint32_t)':                                                                                                                                                                                                               
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h:2517:21: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]                                                                                                                                                                                                             
 2517 |   register uint32_t timeout_cpu_cycles = ADC_LINEARITY_BIT_TOGGLE_TIMEOUT;                                                                                                                                                                                                                                                                                                          
      |                     ^~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                                                                                                                              
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h: In function 'uint32_t LL_ADC_GetCalibrationLinearFactor(ADC_TypeDef*, uint32_t)':                                                                                                                                                                                                                     
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h:2544:21: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]                                                                                                                                                                                                             
 2544 |   register uint32_t timeout_cpu_cycles = ADC_LINEARITY_BIT_TOGGLE_TIMEOUT;                                                                                                                                                                                                                                                                                                          
      |                     ^~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                                                                                                                              
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h: In function 'void LL_ADC_SetOffset(ADC_TypeDef*, uint32_t, uint32_t, uint32_t)':                                                                                                                                                                                                                      
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h:2797:27: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]                                                                                                                                                                                                             
 2797 |   register __IO uint32_t *preg = __ADC_PTR_REG_OFFSET(ADCx->OFR1, Offsety);                                                                                                                                                                                                                                                                                                         
      |                           ^~~~                                                                                                                                                                                                                                                                                                                                                      
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h: In function 'uint32_t LL_ADC_GetOffsetChannel(ADC_TypeDef*, uint32_t)':                                                                                                                                                                                                                               
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h:2869:33: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]                                                                                                                                                                                                             
 2869 |   register const __IO uint32_t *preg = __ADC_PTR_REG_OFFSET(ADCx->OFR1, Offsety);                                                                                                                                                                                                                                                                                                   
      |                                 ^~~~
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h: In function 'uint32_t LL_ADC_GetOffsetLevel(ADC_TypeDef*, uint32_t)':
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h:2895:33: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
 2895 |   register const __IO uint32_t *preg = __ADC_PTR_REG_OFFSET(ADCx->OFR1, Offsety);
      |                                 ^~~~
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h: In function 'void LL_ADC_SetOffsetSignedSaturation(ADC_TypeDef*, uint32_t, uint32_t)':
[... cut ...]
Libraries/STM32Cube_FW_H7_V1.5.0/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_adc.h:6213:33: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
 6213 |   register const __IO uint32_t *preg = __ADC_PTR_REG_OFFSET(ADCx->JDR1, ((Rank & ADC_INJ_JDRX_REGOFFSET_MASK) >> ADC_JDRX_REGOFFSET_POS));
```

The same is possibly true for other header files, also in HAL code for other STM32 CPUs, for example here:

https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_ll_adc.h#L1828

I think, the keyword should just be removed.

ASELSTM, 2020-02-03:
Hi thomask77,

Thank you for this one other issue you reported. I will forward it to our development team for analysis. I will let you updated as soon as I get an answer.

With regards,

ASELSTM, 2020-02-04:
Hi thomask77,

Please allow me to introduce @PCHSTM from our development teams who can better treat your issue.

With regards

ASELSTM, 2020-04-01:
Hi Thomask77,

The issue you pointed out has been confirmed by our development teams. This will be fixed and made available in a future firmware package release. 

Meanwhile,they suggest you to replace the key word `register` by  `__IO ` to avoid this issue.

Thank you once more for your contribution.

With regards,

ASELSTM, 2020-04-02:
ST Internal Reference: 83751

closed 2020-08-03 by ALABSTM