STMicroelectronics/STM32CubeH7 #17: Issue in USB implementation

engycz, 2020-01-20:
Hello.
I have found an issue in USB implementation in STM32CubeH7 (1.4.0 - 1.6.0). Steps to reproduce:
1) Open the project from STM32H743I-EVAL\Applications\USB_Device\MSC_Standalone\MDK-ARM directory
2) Apply HEAP fix from https://github.com/STMicroelectronics/STM32CubeH7/pull/16
3) Build the project and download it to STM32H743I-EVAL board
4) Wait for STM driver letter to appear
5) Copy a file with approx size 20MB to STM drive
6) Start copying above file from STM drive
7) In the middle of copying unplug the USB cable
8) Plug USB cable back
9) From this time USB is not recognized by the operating system anymore 
[image]

The issue is not present in STM32CubeH7 1.3.2

ALABSTM, 2020-02-03:
Hi Engycz,

Please allow me to introduce @bouattay from our USB development teams. Thank you for having reported this point.

With regards,

engycz, 2020-02-04:
The proposed change from ST Case Number 00097303 to stm32h7xx_ll_usb.c fixes this issue

```
HAL_StatusTypeDef USB_DeactivateEndpoint(USB_OTG_GlobalTypeDef *USBx, USB_OTG_EPTypeDef *ep)
{
  uint32_t USBx_BASE = (uint32_t)USBx;
  uint32_t epnum = (uint32_t)ep->num;

  /* Read DEPCTLn register */
  if (ep->is_in == 1U)
  {
    USBx_DEVICE->DEACHMSK &= ~(USB_OTG_DAINTMSK_IEPM & (uint32_t)(1UL << (ep->num & EP_ADDR_MSK)));
    USBx_DEVICE->DAINTMSK &= ~(USB_OTG_DAINTMSK_IEPM & (uint32_t)(1UL << (ep->num & EP_ADDR_MSK)));
-->    if (USBx_INEP(epnum)->DIEPCTL & USB_OTG_DIEPCTL_EPENA)
-->    {
-->      USBx_INEP(epnum)->DIEPCTL |= USB_OTG_DIEPCTL_SNAK;
-->      USBx_INEP(epnum)->DIEPCTL |= USB_OTG_DIEPCTL_EPDIS;
-->    }
    USBx_INEP(epnum)->DIEPCTL &= ~(USB_OTG_DIEPCTL_USBAEP |
                                   USB_OTG_DIEPCTL_MPSIZ |
                                   USB_OTG_DIEPCTL_TXFNUM |
                                   USB_OTG_DIEPCTL_SD0PID_SEVNFRM |
                                   USB_OTG_DIEPCTL_EPTYP);
  }
  else
  {
    USBx_DEVICE->DEACHMSK &= ~(USB_OTG_DAINTMSK_OEPM & ((uint32_t)(1UL << (ep->num & EP_ADDR_MSK)) << 16));
    USBx_DEVICE->DAINTMSK &= ~(USB_OTG_DAINTMSK_OEPM & ((uint32_t)(1UL << (ep->num & EP_ADDR_MSK)) << 16));
-->    if (USBx_OUTEP(epnum)->DOEPCTL & USB_OTG_DOEPCTL_EPENA)
-->    {
-->      USBx_OUTEP(epnum)->DOEPCTL |= USB_OTG_DOEPCTL_SNAK;
-->      USBx_OUTEP(epnum)->DOEPCTL |= USB_OTG_DOEPCTL_EPDIS;
-->    }
    USBx_OUTEP(epnum)->DOEPCTL &= ~(USB_OTG_DOEPCTL_USBAEP |
                                    USB_OTG_DOEPCTL_MPSIZ |
                                    USB_OTG_DOEPCTL_SD0PID_SEVNFRM |
                                    USB_OTG_DOEPCTL_EPTYP);
  }

  return HAL_OK;
}
```

bouattay, 2020-02-05:
Hello Engycz,

The fix is internally implemented and should be visible for H7 family starting from Cube Release v1.8.0

Regards,
Aymen

closed 2020-08-03 by ALABSTM