STMicroelectronics/STM32CubeH7 #18: Invalid HAL version number in stm32h7xx_hal.c file

engycz, 2020-01-20:
There is an invalid HAL version number in stm32h7xx_hal.c file.
HAL version 1.6.0 is reported as 1.7.0
https://github.com/STMicroelectronics/STM32CubeH7/blob/3d3ffbe6e15fc7e4d8a4d275e9fb3a334e31c8a3/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal.c#L53

ALABSTM, 2020-02-04:
Hi Jiří,

Actually, STM32CubeH7 firmware package v1.6.0 includes HAL package v1.7.0, as indicated within the **Release_Notes.html** file.

Please allow me to introduce @BJASTM from our development teams who can better answer your question.

With regards,

engycz, 2020-02-04:
It's confusing that the numbering differs, isn't it?

ALABSTM, 2020-03-30:
Hi Jiří,

I asked our development teams for an explanation about this gap between the firmware's and the HAL's versions, particularly as the HAL's precedes the firmware's.

This gap started from the STM32CubeH7 **v1.4.0** published on the 5th of April 2019. In the frame of this publication, CMSIS Device and HAL components were already versioned v1.5.0 both.

Actually, at some point in the past, our development teams have been requested to deliver a major release of CMSIS Device and HAL components **outside the frame of** a firmware package. This has been done on the 30th of November 2018 when version **v1.4.0** of both components has been released.

To summarize, we have:
* CMSIS D + HAL v1.3.0 --> FW v1.3.0
* CMSIS D + HAL v1.4.0 (released as **standalone** components)
* CMSIS D + HAL v1.5.0 --> FW v1.4.0
* CMSIS D + HAL v1.6.0 --> FW v1.5.0
* CMSIS D + HAL v1.7.0 --> FW v1.6.0
* CMSIS D + HAL v1.8.0 --> FW v1.7.0

Below are links to the different release notes in case you would like to have an overview of the different publication dates:
* Release note of the STM32CubeH7 FW package ([link](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Release_Notes.html))
* Release note of the H7 CMSIS Device component ([link](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/CMSIS/Device/ST/STM32H7xx/Release_Notes.html))
* Release note of the H7 HAL driver component ([link](https://htmlpreview.github.io/?https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Release_Notes.html))

I hope this answers your question and excuse the delayed reply.

With regards,

**PS:** You may notice that in the firmware's release note, version 1.7.0 of the firmware is still reported to integrate version v1.7.0 of both CMSIS Device and HAL components. This is simply a typo that should be corrected soon hopefully.

ALABSTM, 2020-04-02:
Hi Jiří,

I hope you find in my last comment a satisfying answer to your question. Please allow me to close this issue now.

With regards,

closed 2020-04-02 by ALABSTM