STMicroelectronics/STM32CubeH7 #19: __HAL_RCC_AHB3_FORCE_RESET() sometime causes microprocessor to reset

david-oz00, 2020-01-24:
**Setup**
Hardware: Proprietary hardware containing STM32H743 and external FLASH and external SDRAM.
IDE: VisualGDB with GCC 9.2.1

**Bug Description**
When the HAL_DeInit() function calls the __HAL_RCC_AHB3_FORCE_RESET() function the microprocessor sometimes resets.

    #define __HAL_RCC_AHB3_FORCE_RESET()          (RCC->AHB3RSTR = 0xFFFFFFFFU)

**How To Reproduce**
1. Indicate the global behavior of your application project
A small FLASH programmer is running in internal RAM (DTCMRAM @ 0x20000000).  After it programs the external flash it unloads itself and calls HAL_DeInit() which calls __HAL_RCC_AHB3_FORCE_RESET().  At this point the microprocessor resets.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)
The __HAL_RCC_AHB3_FORCE_RESET() resides in stm32h7xx_hal_rcc.h

3. The use case that generates the problem
Calling the HAL_DeInit() when the program prepares to unload itself.

4. How we can reproduce the problem
I don't know all conditions which makes the function fail.  I suspect it has something to do with calling the function from internal RAM.  I don't think you need to reproduce the problem.  It is clearly a bug (see below) which should be rectified.

**Additional context**
The code writes all 1's to the AHB3RSTR register.  However, the STM32H7 reference manual says the reserved bits in that register must remain at the reset value (i.e. 0's)

This code:

    #define __HAL_RCC_AHB3_FORCE_RESET()          (RCC->AHB3RSTR = 0xFFFFFFFFU)

should be replaced by:

    #define __HAL_RCC_AHB3_RESET_ALL               (RCC_AHB3RSTR_MDMARST |       \
                                                    RCC_AHB3RSTR_DMA2DRST |      \
                                                    RCC_AHB3RSTR_JPGDECRST |     \
                                                    RCC_AHB3RSTR_FMCRST |        \
                                                    RCC_AHB3RSTR_QSPIRST |       \
                                                    RCC_AHB3RSTR_SDMMC1RST)

    #define __HAL_RCC_AHB3_FORCE_RESET()          (RCC->AHB3RSTR = __HAL_RCC_AHB3_RESET_ALL)    

**Screenshots**
None

david-oz00, 2020-01-24:
I've noticed similar functions incorrectly write all 1's to the registers.  E.g

    #define __HAL_RCC_AHB1_FORCE_RESET()             (RCC->AHB1RSTR = 0xFFFFFFFFU)
    #define __HAL_RCC_AHB2_FORCE_RESET()           (RCC->AHB2RSTR = 0xFFFFFFFFU)
    #define __HAL_RCC_APB3_FORCE_RESET()         (RCC->APB3RSTR = 0xFFFFFFFFU)
    #define __HAL_RCC_APB1L_FORCE_RESET()        (RCC->APB1LRSTR = 0xFFFFFFFFU)
    #define __HAL_RCC_APB1H_FORCE_RESET()        (RCC->APB1HRSTR = 0xFFFFFFFFU)
    #define __HAL_RCC_APB2_FORCE_RESET()         (RCC->APB2RSTR = 0xFFFFFFFFU)
    #define __HAL_RCC_APB4_FORCE_RESET()           (RCC->APB4RSTR = 0xFFFFFFFFU)

I haven't seen any detrimental affects from these functions, unlike the one described in the initial report.  However, technically the HAL should not be writing 1's to the reserved bits of these registers.

ALABSTM, 2020-02-03:
Hi David,

Thank you for this report. May I ask you the STM32CubeH7 firmware revision you have used to get this problem?

With regards,

david-oz00, 2020-02-04:
Hi ALABSTM,
The package I am using is STM32Cube FW_H7 V1.6.0.  I've also attached an excerpt from the stm32H7 reference manual stating the reserve bits should not change.

[image]

ALABSTM, 2020-02-04:
Hi David,

May I ask you why you have closed this issue? It seems to be relevant.

With regards,

david-oz00, 2020-02-04:
I was about to ask you the same thing :)  I think it is relevant too.  Can you re-open it?

david-oz00, 2020-02-04:
I must have hit the close button by mistake.  Sorry.

ALABSTM, 2020-02-04:
Hi David,

Done! I will forward it to our development teams. I will be back to you as soon as I get an answer.

With regards,

ALABSTM, 2020-02-04:
Hi David,

Please allow me to introduce @BJASTM from our development teams who can better answer your question.

With regards,

ALABSTM, 2020-02-07:
Hi David,

Our development teams confirmed the issue you have pointed out. This will be fixed and made available in a future firmware package release. Thank you once more for your contribution.

With regards,

ALABSTM, 2020-02-07:
ST Internal Reference: 80761

veda-s4dhak, 2020-05-19:
@ALABSTM, I see that this is on the "To Release" section on the internal bug tracker on the development board. Is this available now? When can we expect this?

ALABSTM, 2020-05-26:
Hi @veda-s4dhak,

The fix has been internally implemented but not is released yet. It is planned to be part of the **v1.8.0** firmware package release. We cannot share a date for the moment unfortunately, but we hope it will be soon enough.

In the meanwhile, please find attached [this](https://github.com/STMicroelectronics/STM32CubeH7/files/4679968/stm32h7xx_hal_rcc.zip) patch with the updated `stm32h7xx_hal_rcc.h` file.

With regards,

closed 2020-08-03 by ALABSTM