STMicroelectronics/STM32CubeH7 #20: The HAL_NOR_ProgramBuffer() has a few defects

david-oz00, 2020-01-24:
**Describe the set-up**
Hardware: Proprietary hardware containing STM32H743 and external FLASH and external SDRAM.
IDE: VisualGDB with GCC 9.2.1

**Describe the bug**
1.  The calculation for *p_endaddress* is incorrect.  It adds *uwBufferSize* to *uwAddress* to create the pointer, but *uwBufferSize* is the word count, not the byte count.  Therefore *uwBufferSize* should be multiplied by 2 to give the byte count.  Alternatively, *uwAddress* could be turned into a word pointer prior to adding *uwBufferSize* to it.
2. The local variables *p_currentaddress* and *lastloadedaddress* are used incorrectly in some parts of the code.  The variables contain the **internal** address of the FLASH, but they are sometimes used as if they are offsets from the device location.  When they are used as pointers to the FLASH they should have the FLASH device address added to them beforehand.

**How To Reproduce**
1. Indicate the global behavior of your application project
The bootloader is writing an application to external FLASH.

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)
The *HAL_NOR_ProgramBuffer()* function in the file **stm32h7xx_hal_nor.c**

3. The use case that generates the problem
Write a buffer to the external NOR FLASH.

4. How we can reproduce the problem
Call it.

**Additional context**
The whole function needs to be cleaned up.  I would replace *p_endaddress* with a count down counter.  However, if you are just interested in fixing it:

The code:

    p_endaddress      = (const uint16_t *)(uwAddress + (uwBufferSize - 1U));
Should be changed to:

    p_endaddress      = (const uint16_t *)(uwAddress + (2 * (uwBufferSize - 1U)));
or alternatively

    p_endaddress      = ((const uint16_t *)uwAddress) + (uwBufferSize - 1U);

Later in the function, the code:

    while (p_currentaddress <= p_endaddress)
    {
      /* Store last loaded address & data value (for polling) */
      lastloadedaddress = (uint32_t)p_currentaddress;

      NOR_WRITE(p_currentaddress, *data);

      data++;
      p_currentaddress ++;
    }

    NOR_WRITE((uint32_t)(lastloadedaddress), NOR_CMD_DATA_BUFFER_AND_PROG_CONFIRM);

Should be changed to:

    while (p_currentaddress <= p_endaddress)
    {
      /* Store last loaded address & data value (for polling) */
      lastloadedaddress = (uint32_t)p_currentaddress;

      NOR_WRITE(deviceaddress + (uint32_t)p_currentaddress, *data);

      data++;
      p_currentaddress ++;
    }

      NOR_WRITE((uint32_t)(deviceaddress + lastloadedaddress), NOR_CMD_DATA_BUFFER_AND_PROG_CONFIRM);

**Screenshots**
None

ASELSTM, 2020-02-03:
Hi David,

Thank you for reporting this issue. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get an answer. Thank you in advance for your patience.

With regards,

ASELSTM, 2020-02-04:
Hi David, 

As promised i have reported your remarks to our development teams and they confirmed the issue. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-02-04:
ST Internal Reference: 80491

AAganichev, 2020-08-04:
Hi!

I'd suggest to unify API for the whole set of HAL_NOR_* functions. I'd say HAL_NOR_Read and HAL_NOR_Write should accept uwAddress relative to base address of Flash instead of absolute pointer pAddress.

And there's another issue with the CubeMX: if you select different FMC bank mappings the addresses are not changed respectively. It would be nice to be fixed as well.

Should I report these issues separately or they can be done within same issue?

Regards,
Alexander

ASELSTM, 2021-03-09:
Hi @david-oz00,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.9.0 of the STM32CubeH7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-09 by ASELSTM