STMicroelectronics/STM32CubeH7 #21: SD-Card: hard-coded clock dividers only work for 200MHz MMCCLK

thomask77, 2020-01-24:
For MMC and SD-Card operation, the clock dividers are hard-coded and work for 200 MHz `sdmmc_clk` _only_:

https://github.com/STMicroelectronics/STM32CubeH7/blob/3d3ffbe6e15fc7e4d8a4d275e9fb3a334e31c8a3/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_ll_sdmmc.h#L697-L704

CubeMX5 allows to configure the SDMMC peripheral with up to 480 MHz, so the clocks will be much to fast (e.g. 960 kHz during init instead of 400 kHz).

-> The dividers should be re-calculated according to the peripheral clock in  `stm32h7xx_hal_sd.c` and `stm32h7xx_hal_mmc.c`.

(`stm32h7xx_hal_sd.c` already uses `HAL_RCCEx_GetPeriphCLKFreq(RCC_PERIPHCLK_SDMMC)` to calculate some delays. This must also be done for the divisors).

ASELSTM, 2020-02-03:
Hi thomask77,

The issue you pointed out has been communicated to our development teams. It will be investigated soon. We will be back to you as soon as we get an answer. Thank you for your patience.

With regards,

ASELSTM, 2020-02-14:
Hi thomask77,

Our development team  has confirmed the issue you pointed out and here is their reply :

The clock divider is configured through the init structure to allow the user setting the desired clock frequency. So, the user could use either the define provided in the .h file or define his own dividers.
But since the high speed and ultra-high speed features have been implemented, these define are also used in the driver. Then, as indicated in the issue, to have the appropriate clock frequency when not running at max frequency, it should be necessary to calculate the correct divider and not use directly the define.

Since the issue is confirmed, a fix will be implemented and made available in future release. Thank you once again for pointing out the issue and for your active contribution.

With regards,

ASELSTM, 2020-02-14:
ST Internal Reference: 80503

ASELSTM, 2021-03-09:
Hi @thomask77,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.9.0 of the STM32CubeH7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-09 by ASELSTM

davhak, 2021-10-16:
Sorry if this is not correct but i do not see any difference in content of (stm32h7xx_ll_sdmmc.h) of version v1.9.0 and of the same file in v.1.8.3. Basically, all the HAL Driver files in version v1.9.0 are identical to previous version v1.8.3. So it is not clear what has been fixed in terms of "SDMMC_INIT_CLK_DIV" macro. In STM32H& the M4 core can be as high as 240 MHz. and the divider value of 0xFA still needs to be a different number.