STMicroelectronics/STM32CubeH7 #22: Stalling at the DataOutStage in usbd_core.c leads to enumeration issues under Windows 10/7

fstap, 2020-01-30:
Hi,

enumeration fails for a lot of users because of stalling of the endpoint during the usb configuration stage. The issue seems to occure with pretty much every STM32, I can confirm it for the STM32L4, STM32H7 and STM32F7 series. The effects can easily be seen by using a USB protocol analyzer like USBTrace. A quick fix is to remove the line USBD_LL_StallEP(pdev, 0U); completely which does not interfere with the USB functionality in any way. The folowing thread describes the bug in more detail where I also recently posted the solution: https://community.st.com/s/question/0D50X00009XkiIASAZ/stm32f405rg-vcp-error-this-device-cannot-startcode-10

Kind regards

fstap

ALABSTM, 2020-02-03:
Hi Fstap,

Thank you for having reported this issue.

This issue may already have been detected and fixed internally. If it is the case, the fix may already be available in some firmware Cube package recently published. If not, we will make a fix available soon, hopefully.

Please allow me to introduce @bouattay who will comment your post and provide you with more details about this point and thank you once more for your contribution.

With regards,

bouattay, 2020-02-03:
Hi Fstap,

Absolutely this is known issue already fixed in our latest USB device Lib v2.6.0 to be published sooner 
 proposed fix:
usbd_core.c:
USBD_LL_DataOutStage()
{
...
    else
    {
#if 0
      if (pdev->ep0_state == USBD_EP0_STATUS_OUT)
      {
        /*
         * STATUS PHASE completed, update ep0_state to idle
         */
        pdev->ep0_state = USBD_EP0_IDLE;
        (void)USBD_LL_StallEP(pdev, 0U);
      }
#endif
    }
}

USBD_LL_DataInStage()
{
...
    else
    {
#if 0
      if ((pdev->ep0_state == USBD_EP0_STATUS_IN) ||
          (pdev->ep0_state == USBD_EP0_IDLE))
      {
        (void)USBD_LL_StallEP(pdev, 0x80U);
      }
#endif
    }
}

Regards,
Aymen

closed 2020-08-03 by ALABSTM