STMicroelectronics/STM32CubeH7 #24: HAL_RTC_Init() looses time

thomask77, 2020-02-10:
**Describe the set-up**
I'm using a STM32H743 on a custom board with 32768 Hz LSE crystal and battery-backed up RTC.

**Describe the bug**
Every time `HAL_RTC_Init()` is called, the RTC looses some sub-seconds.

`HAL_RTC_Init` should check if the RTC is already initializied (year != 0) and/or matches the desired configuration, and skip unnecessary stopping of the clock. According to the data sheet 45.3.8, the library should also check the `INITS` flag in the RTC_ISR register.

This is big a problem, because CubeMX5 will emit code that calls `HAL_RTC_Init` on each reset, without checking if it's already initialized. On each reset, you will loose up to 1s.

**How To Reproduce**
1. Read time with `HAL_RTC_GetTime` &`HAL_RTC_GetDate` and print to console
2. Call `HAL_RTC_Init`
3. Repeat

While doing this, the clock will gradually loose seconds.

ALABSTM, 2020-02-12:
Hi Thomask77,

Thank you for this one more contribution. This issue has also been reported by the ST Community. We are already working on it. We will let you updated as soon as there is something new from our development teams.

With regards,

ALABSTM, 2020-02-12:
ST Internal Reference: 45013

ALABSTM, 2020-03-13:
Hi Thomask77,

According to our development teams, the issue is at MX initialization-level not at HAL one. The suggested solution looks like the following:
* At user application-level, RTC initialization shall be conditioned:
```diff
+/* Do not re-initialize RTC in case it has already been to avoid loosing seconds at each reset */
+if ((RCC->BDCR & RCC_BDCR_RTCEN) == 0)
+{
  if(HAL_RTC_Init(&RtcHandle) != HAL_OK)
  {
    /* Initialization Error */
    Error_Handler(); 
  }
+}
+else
+{
+  HAL_PWR_EnableBkUpAccess();
+}
```
* These line of codes shall be added into the main() function of your example or application (as the one below, for instance):
https://github.com/STMicroelectronics/STM32CubeH7/blob/3d3ffbe6e15fc7e4d8a4d275e9fb3a334e31c8a3/Projects/NUCLEO-H743ZI/Examples/RTC/RTC_Alarm/Src/main.c#L95-L99

We hope these details help you solve the issue from your side.

This issue has been reported to the CubeMX team. A fix will be available in a future CubeMX release. Unfortunately, we cannot share a date for the moment.

As this is a **CubeMX-related** issue and not directly related to the firmware published whithin this repository, please allow me to close it. Thank you for your comprehension and thank you again for your contribution.

With regards,

ALABSTM, 2022-04-13:
Hi @thomask77,

After further discussions with our development teams, it has been agreed this point could be handled at driver's level. Hence, I will reopen this issue pending a fix is published.

With regards,

ALABSTM, 2022-04-13:
Hi @thomask77,

As for the fix, it will consist in checking the `RTC_ISR_INITS` bit, which, as described in the reference manual:
* is equal to `0` if the calendar has **not** been initialized
* is equal to `1` if the calendar has been initialized

Thus, the fix would look something like this:
```diff
HAL_StatusTypeDef HAL_RTC_Init(RTC_HandleTypeDef *hrtc)
{
  ...
+    /* Check whether the calendar needs to be initialized */
+    if (RTC_ISR_INITS == 0U)
+    {
       /* Disable the write protection for RTC registers */
       __HAL_RTC_WRITEPROTECTION_DISABLE(hrtc);
       
       /* Enter Initialization mode */
       status = RTC_EnterInitMode(hrtc);
       ...
+    }
}
```

With regards,

Judicaelkouadio, 2022-04-13:
Hi,

unfortunately when use the code:
+/* Do not re-initialize RTC in case it has already been to avoid loosing seconds at each reset */
+if ((RCC->BDCR & RCC_BDCR_RTCEN) == 0)
+{
  if(HAL_RTC_Init(&RtcHandle) != HAL_OK)
  {
    /* Initialization Error */
    Error_Handler(); 
  }
+}
+else
+{
+  HAL_PWR_EnableBkUpAccess();
+}

I always get zero when I try to read time.

Judicaelkouadio, 2022-04-13:
Hi,

when I apply the last solution, the function getTime sometimes gets 00:00:00.

Any idea?

Judicaelkouadio, 2022-04-13:
Hi,

Thanks for the solution, it works but the function getTime sometimes retuns 00:00:00.

Any idea?

PG430, 2022-04-14:
Hi,

Thanks for providing solution. Applied same solution but still losing 1 second. 
every 5 second printing "Wakeup..." string on terminal and soft reset on every 1 minute
Logs are attached. 

[image]

ALABSTM, 2022-04-15:
Hi @Judicaelkouadio and @PG430,

Do you confirm you implemented the fix in `hal_rtc.c` rather than in `main.c`? I just edited my previous comment to emphasize this important point.

Please keep me informed.

With regards,

Judicaelkouadio, 2022-04-15:
@ALABSTM 
I confirm that.
The problem has been solved by:
__HAL_RTC_WRITEPROTECTION_DISABLE(&hrtc);
  if (HAL_RTC_WaitForSynchro(&hrtc) != HAL_OK)
  {
	  Error_Handler();
  }
  __HAL_RTC_WRITEPROTECTION_ENABLE(&hrtc);

before the fonctions getDate and getTime.

Thanks.

PG430, 2022-04-18:
hi,

I've made your suggested changes in stm32wlxx_hal_rtc.c. pls check attached snapshot of the source

[image]

PG430, 2022-05-04:
@ASELSTM Please share any available update regarding this concern.

ALABSTM, 2022-07-29:
Hi @PG430,

We are deploying the fix to the HAL RTC drivers of the different STM32Cube firmware packages. Hopefully, the fix will be integrated into the STM32CubeH7 firmware soon.

With regards,

PG430, 2022-11-07:
@ALABSTM  is this issue also fixed for STM32WLE5JC?

PG430, 2022-12-08:
@ALABSTM we are working on new release of STM32WL "STM32Cube_FW_WL_V1.3.0". release has introduced changes in RTC driver which we believe is fixed for this issue(RTC loosing time issue). After running RTC overnight we have seen issues is still persist. 

Test Start time : 17:55:56, UTC time 1670415956   
Test End Time:  9:48:00, UTC time 1670473080,     (57124 Seconds)
Difference of  UTC time should be : 57124 (1670473080 - 1670415956)

But, From attached logs we can see Test End time is 9:48:00 and total seconds are 57135s837 (check last line in logs). 
RTC had generated  57135s837 seconds instead of  57124 Seconds.

So, 57135 - 57124 = 11 Seconds are increased in actual time

kaunomedis, 2023-01-08:
2023.01.08, 1.8.4 still no fix, RTC still loosing seconds after power up.

CedricRBR, 2023-02-27:
I noticed that other than `HAL_RTC_Init()` any interrupt that gets enabled has its counter reset to zero in `MX_RTC_Init()`.
For me it was the call to `HAL_RTCEx_WakeUpTimer_IT()`. 
I carved the function and kept only what was necessary and called those functions in `MX_RTC_Init()` instead of the call to `HAL_RTCEx_WakeUpTimer_IT()`

ALABSTM, 2023-04-04:
Hi,

Issue fixed in version **1.11.0** of the STM32cubeH7 firmware.

With regards,

closed 2023-04-04 by ALABSTM