STMicroelectronics/STM32CubeH7 #25: Wrong timer identifiers on some HRTIM functions doc

higaski, 2020-02-14:
The HRTIM header contains the following definitions as timer identifiers
`/** @defgroup HRTIM_Timer_identifier HRTIM Timer identifier
  * @{
  * @brief Constants defining timer identifiers
  */
#define HRTIM_TIMERID_MASTER  (HRTIM_MCR_MCEN)   /*!< Master identifier  */
#define HRTIM_TIMERID_TIMER_A (HRTIM_MCR_TACEN)  /*!< Timer A identifier */
#define HRTIM_TIMERID_TIMER_B (HRTIM_MCR_TBCEN)  /*!< Timer B identifier */
#define HRTIM_TIMERID_TIMER_C (HRTIM_MCR_TCCEN)  /*!< Timer C identifier */
#define HRTIM_TIMERID_TIMER_D (HRTIM_MCR_TDCEN)  /*!< Timer D identifier */
#define HRTIM_TIMERID_TIMER_E (HRTIM_MCR_TECEN)  /*!< Timer E identifier */`

Some HRTIM functions (e.g. HAL_HRTIM_WaveformCountStop) incorrectly have those definitions documented as
`  * @param  Timers Timer counter(s) to stop
  *                   This parameter can be any combination of the following values:
  *                   @arg HRTIM_TIMERID_MASTER
  *                   @arg HRTIM_TIMERID_A
  *                   @arg HRTIM_TIMERID_B
  *                   @arg HRTIM_TIMERID_C
  *                   @arg HRTIM_TIMERID_D
  *                   @arg HRTIM_TIMERID_E`

As far as I can tell those don't exist, not even as legacy definitions.

ALABSTM, 2020-03-10:
Hi Higaski,

Thank you for having pointed out this issue. Indeed, timer identifiers listed in the header of some HAL functions are not aligned with the actual constant names used.

This will be updated in made available in a future release. Thank you again for your contribution.

With regards,

ALABSTM, 2020-03-10:
ST Internal Reference: 82539

closed 2020-08-03 by ALABSTM