STMicroelectronics/STM32CubeH7 #28: Cannot read Audio Data(MP34DT05TR Mic) based on DFSDM

zhoujiazhao, 2020-03-05:
- **Board**: STM32H747I-DISO
- **SDK**: STM32Cube_FW_H7_V1.6.0
- **Pin**: PD3(CLK), PC7(DATAIN3)

- **Problem**: Filter->DMA bufer OK. **HAL_DFSDM_FilterRegConvCpltCallback** and **HAL_DFSDM_FilterRegConvHalfCpltCallback** function enter normally, but the data is wrong(always the same number). 
I monitor the DFSDM registers, the **RDATAR** register no change.

- **Header file**

```

/* DFSDM Configuration defines */
#define AUDIO_DFSDMx_RIGHT_CHANNEL      DFSDM_CHANNEL_2
#define AUDIO_DFSDMx_LEFT_CHANNEL       DFSDM_CHANNEL_3

#define AUDIO_DFSDMx_LEFT_FILTER        DFSDM1_Filter0
#define AUDIO_DFSDMx_RIGHT_FILTER       DFSDM1_Filter1

   
#define AUDIO_DFSDMx_CLK_ENABLE()                       __HAL_RCC_DFSDM1_CLK_ENABLE()
#define AUDIO_DFSDMx_CKOUT_PIN                          GPIO_PIN_3
#define AUDIO_DFSDMx_CKOUT_DMIC_GPIO_PORT               GPIOD
#define AUDIO_DFSDMx_CKOUT_AF                           GPIO_AF3_DFSDM1
#define AUDIO_DFSDMx_CKOUT_DMIC_GPIO_CLK_ENABLE()       __HAL_RCC_GPIOD_CLK_ENABLE()

#define AUDIO_DFSDMx_DMIC_DATIN_PIN                     GPIO_PIN_7
#define AUDIO_DFSDMx_DMIC_DATIN_GPIO_PORT               GPIOC
#define AUDIO_DFSDMx_DMIC_DATIN_AF                      GPIO_AF4_DFSDM1
#define AUDIO_DFSDMx_DMIC_DATIN_GPIO_CLK_ENABLE()       __HAL_RCC_GPIOC_CLK_ENABLE()
    
/* DFSDM DMA Right and Left channels definitions */
#define AUDIO_DFSDMx_DMAx_CLK_ENABLE()          __HAL_RCC_DMA2_CLK_ENABLE()

#define AUDIO_DFSDMx_DMAx_LEFT_REQUEST          DMA_REQUEST_DFSDM1_FLT0
#define AUDIO_DFSDMx_DMAx_RIGHT_REQUEST         DMA_REQUEST_DFSDM1_FLT1

#define AUDIO_DFSDMx_DMAx_PERIPH_DATA_SIZE      DMA_PDATAALIGN_WORD
#define AUDIO_DFSDMx_DMAx_MEM_DATA_SIZE         DMA_MDATAALIGN_WORD

#define AUDIO_DFSDMx_DMAx_LEFT_STREAM           DMA2_Stream3
#define AUDIO_DFSDMx_DMAx_LEFT_IRQ              DMA2_Stream3_IRQn   
#define AUDIO_DFSDMx_DMAx_LEFT_IRQHandler       DMA2_Stream3_IRQHandler

#define AUDIO_DFSDMx_DMAx_RIGHT_STREAM          DMA2_Stream2
#define AUDIO_DFSDMx_DMAx_RIGHT_IRQ             DMA2_Stream2_IRQn
#define AUDIO_DFSDMx_DMAx_RIGHT_IRQHandler      DMA2_Stream2_IRQHandler

/* Select the interrupt preemption priority and subpriority for the DMA interrupt */
#define AUDIO_IRQ_PREPRIO                    ((uint32_t)0x0E)
```

- **Source file**
```
#include "main.h"

#define SaturaLH(N, L, H) (((N) < (L)) ? (L) : (((N) > (H)) ? (H) : (N)))

/* Private variables ---------------------------------------------------------*/
DFSDM_Channel_HandleTypeDef DfsdmLeftChannelHandle;
DFSDM_Channel_HandleTypeDef DfsdmRightChannelHandle;
DFSDM_Filter_HandleTypeDef DfsdmLeftFilterHandle;
DFSDM_Filter_HandleTypeDef DfsdmRightFilterHandle;

DMA_HandleTypeDef DfsdmLeftDmaHandle;
DMA_HandleTypeDef DfsdmRightDmaHandle;

AUDIO_Drv_t *Audio_Drv = NULL;
WM8994_Init_t codec_init;

/*Buffer location and size should aligned to cache line size (32 bytes) */
[... 407 lines cut ...]
  //AudioRecordCount =  AUDIO_RECORD_MAX_TIME_MS;

  while (1)
  {
    event = osMessageGet(AudioRecordEvent, osWaitForever); //osWaitForever
    if (event.status == osEventMessage)
    {
      switch (event.value.v)
      {
      case AUDIO_RECORD_MESSAGE_EVENT_PROCESS:
        printf("AUDIO_RECORD_MESSAGE_EVENT_PROCESS\r\n");
        break;

      case AUDIO_RECORD_MESSAGE_EVENT_STOP:
        Audio_Record_Stop();
        break;

      default:
        break;
      }
    }
  }
}

````

ASELSTM, 2020-03-31:
Hi Zhoujiazhao,

Thank you for reporting this issue. It will be forwarded to our development team to check it. You will be updated as soon as the provide us with feedback. Thank you once more.

With regards,

ASELSTM, 2020-04-01:
Hi Zhoujiazhao,

I have checked with our development teams. They mentioned that on STM32H747I-DISO board, there is no microphone connected on the DFSDM. However, a connector CN17, is available to connect it.

Did you make sure you that you have connected a microphone on your STM32H747I-DISO board ?

If you already have connected a microphone on your board, you can refer to the example [DFSDM ](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H747I-EVAL/Examples/DFSDM/DFSDM_AudioRecord) available on  STM32H747I-EVAL (which has a connected microphone on DFSDM), it may help you. 

Best regards,

Ibrahfaye, 2020-07-06:
Hello,
Hope this thread still in live? Because i have the same issue. I plug board on cn17 of stm32h747i-disco but there no sound output on headphone. The PlayBuf always have -32768. I layed my work on example from stm32h747-eval DFSDM. Have an idea? Thanks,

ASELSTM, 2020-07-13:
Hi @Ibrahfaye,

Did you make sure that you have connected a microphone on your STM32H747I-DISO ? 

Best regards

Ibrahfaye, 2020-07-14:
Hello ASELSTM,
i'm 100% sure! I plug mics on cn17 inputs. This mics working great when I run Audioweaver software from your Partner DSPConcept.
Regards,

ASELSTM, 2020-08-03:
Hi @Ibrahfaye,

We are very pleased with the interest you show to our products. However, this public organization on GitHub is intended to promote co-development and issues posted here shall exclusively be related to the software published within the different repositories.

For any other issue (e.g. related to documentation, to hardware, to ecosystem, to user application, etc.), you shall address your questions to the [ST Community](https://community.st.com/s/topiccatalog). There you shall look for the adequate topic and submit your question or remark.

You may also, as an alternative solution, use ST On-Line support channel. For that you have to : 

- Go to the [STM32H747I-DISCO](https://www.st.com/en/evaluation-tools/stm32h747i-disco.html) page.

- From the "**Support and Community**" tab select "**Support Home**" then "**Support Request -via web form**" . You will be invited to subscribe to myST and then you will get access to a specific form that will drive you to the ST Division FAEs.

Hope this will help you. As for the issue you posted here, allow us to close it now. Thank you for your understanding.

With regards,

closed 2020-08-03 by ASELSTM