STMicroelectronics/STM32CubeH7 #30: Repository is behind the software provided by the STM32CubeH7 distribution

sebhub, 2020-03-19:
I downloaded the STM32CubeH7 from:

https://www.st.com/en/embedded-software/stm32cubeh7.html

In this package the HAL version is: STM32H7xx HAL Driver version number V1.8.0

In this repository I see only V1.7.0 or earlier. Why is this repository here not up to date?

ALABSTM, 2020-04-02:
Hi @sebhub,

Indeed, STM32CubeH7 firmware pacakge v1.7.0 (including HAL driver v1.8.0) has been published days after it was on the st.com website. We do apologize for this delay.

Now, the firmware package is also available on GitHub. Please note also that if you are only interested in the HAL driver, you can directly clone (or pull) [this](https://github.com/STMicroelectronics/stm32h7xx_hal_driver) dedicated repository.

With regards,

ALABSTM, 2020-04-03:
Hi @sebhub,

I hope you find in my previous reply a satisfying answer to your question. Please allow me to close this issue now.

With regards,

closed 2020-04-03 by ALABSTM