STMicroelectronics/STM32CubeH7 #31: USART Receive timeout interrupt not enabled.

david-oz00, 2020-03-29:
**Describe the set-up**
Hardware: Proprietary hardware containing STM32H743 and external FLASH and external SDRAM.
IDE: VisualGDB with GCC 9.2.1
 
**Additional context**
Setting up the UART receive timeout involves calling these functions:
`HAL_UART_ReceiverTimeout_Config(mHuart, mBitsPerCharacter);
HAL_UART_EnableReceiverTimeout(mHuart);`

However, that does not enable the receive timeout **interrupt**.  The timeout detection is enabled, but not the interrupt.  The following line should be added to the 'enable' routine, or placed in a separate call:

`SET_BIT(mHuart->Instance->CR1, USART_CR1_RTOIE);`

Also, after the receive completes, the interrupt should be disabled.

`CLEAR_BIT(mHuart->Instance->CR1, USART_CR1_RTOIE);`

**Screenshots**
None

ASELSTM, 2020-03-31:
Hi David-oz00,

Thank you for this one other issue you reported. I will forward it to our development team for analysis. I will let you updated as soon as I get an answer.

With regards,

ASELSTM, 2020-09-23:
Hi @david-oz00,

As promised we have reported your report to our development teams and they confirmed the issue. A fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-09-23:
ST Internal Reference: 86997

ASELSTM, 2021-03-09:
Hi @david-oz00,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.9.0 of the STM32CubeH7 published recently on GitHub.

Thank you again for having reported.

With regards,