STMicroelectronics/STM32CubeH7 #32: CubeMX: 0-pointer access in generated code if TIM6 is used for HAL tick and DAC is enabled

thomask77, 2020-04-02:
Hi!

I don't know where to report CubeMX bugs, so I leave it here. It might also be considered a HAL problem.

I noticed this behavior after enabling a MPU region to catch 0-pointer accesses.

If TIM6 is used as the HAL Timebase **and** the DAC is enabled, there may be a 0-Pointer access in HAL_DAC_IRQHandler (called by TIM6_DAC_IRQHandler), **if** a tick IRQ happens before MX_DAC1_Init() is called.

The same bug likely exists in all other HAL and CubeMX versions.

Simple workaround: Use another timer, that has a non-shared IRQ handler.

best regards,
Thomas

ASELSTM, 2020-04-09:
Hi @thomask77 ,

Thank you for this report. We tried to reproduce the issue you described, but we couldn't notice the problem. May I ask you if you were able to reproduce it, and if so what is the procedure followed to get it ?

According to a first analysis, we noticed that CubeMx initializes by default the DAC1 before initializing the TIMx instances. Thus, MX_DAC1_Init() is always called before MX_TIM6_Init(). Therefore, no TIM interruption is generated before DAC initialization.

Moreover, the code generated by CubeMx remains just an initialization, so, no Timbase interruption is enabled until the user implicitly activates them either by calling the interrupt mode APIs (such as `HAL_TIM_Base_Start_IT()` ) or by activating the update interruption ( calling `__HAL_TIM_ENABLE_IT(htim, TIM_IT_UPDATE)` or directly enabling `TIM_DIER_UIE` bit).
 
Take care and stay safe.

With regards

thomask77, 2020-04-12:
Yes, that's the problem - `HAL_TIM_Base_Start_IT()` is called!

The call chain is:
- `main()`
- `HAL_Init()`:
  - https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal.c#L134
- `HAL_InitTick()` (weak function, overridden by CubxMX with a file derived from a template contained in this repository):
  - https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_timebase_tim_template.c#L58
- `HAL_TIM_Base_Start_IT()`:
  - https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Src/stm32h7xx_hal_timebase_tim_template.c#L114-L121

This happens before `HAL_DAC_Init()` is called.

Null pointer access is then triggered here:

```c
void TIM6_DAC_IRQHandler(void)
{
  /* USER CODE BEGIN TIM6_DAC_IRQn 0 */

  /* USER CODE END TIM6_DAC_IRQn 0 */
  HAL_DAC_IRQHandler(&hdac1);
  HAL_TIM_IRQHandler(&htim6);
  /* USER CODE BEGIN TIM6_DAC_IRQn 1 */

  /* USER CODE END TIM6_DAC_IRQn 1 */
}
```

`HAL_DAC_IRQHandler` will use the uninitilized `Instance` pointer.

Steps to reproduce:
* Start CubeMX 5.6.0
* Select NUCELO-H743ZI in Board Selector
* Init all peripherals to default mode -> Yes (doesn't matter)
* System Core -> SYS -> Timebase Source: TIM6
* Analog -> DAC1, OUT1 mode: Connected to external pin only

Generate code.

ASELSTM, 2020-04-28:
Hi @thomask77 ,

Our development teams confirmed that this issue is rather related to CubeMX than to the firmware published within this repository. Unfortunately we don't treat aspect related to CubeMx tool in our GitHub repositories.

Please allow me to redirect you to our ST Community. You can report this issue in the [CubeMX related section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx). There, you will hopefully find people from our CubeMX development teams who will take in charge your request.

Regarding the point you reported, our development teams suggest you a few fixes to avoid the 0 pointer access in this case : 
 
- Force the instances when declaring the handles as following :

```
  DAC_HandleTypeDef     hdac1 = {.Instance = DAC1}; 
  TIM_HandleTypeDef     htim6 = {.Instance = TIM6};
```

- Checking each peripheral state within the TIM6_DAC_IRQHandler before calling the corresponding IRQHandler : 

```
void TIM6_DAC_IRQHandler(void)
{
  /* USER CODE BEGIN TIM6_DAC_IRQn 0 */

  /* USER CODE END TIM6_DAC_IRQn 0 */
  if(HAL_DAC_GetState(&hdac1) != HAL_DAC_STATE_RESET)
  {
    HAL_DAC_IRQHandler(&hdac1);
  }
  if(HAL_TIM_GetState(&htim6) != HAL_TIM_STATE_RESET)
  {
    HAL_TIM_IRQHandler(&htim6);
  }
  /* USER CODE BEGIN TIM6_DAC_IRQn 1 */

  /* USER CODE END TIM6_DAC_IRQn 1 */
}
```

Please allow me to close this issue now. Thank you again for your report.

With regards,

closed 2020-04-28 by ASELSTM

thomask77, 2020-04-28:
I assume you have an internal process, where the CubeMX team monitors bugs here.

Unfortunately, I can't spend the time to report it on the forum. I've already solved the problem for me, but it would be good for your product if the bug gets fixed for everyone else.

Best regards,
Thomas!