STMicroelectronics/STM32CubeH7 #33: CubeMX: 0-Pointer access in generated MX_ETH_Init code

thomask77, 2020-04-02:
Hi!

I don't know where to report CubeMX bugs, so I leave it here. It might also be considered a HAL problem.

I noticed this behavior after enabling a MPU region to catch 0-pointer accesses.

CubeMX 5.6.0 generates this code for Ethernet initialization:
```c
static void MX_ETH_Init(void)
{
...
  heth.Instance = ETH;
  heth.Init.MACAddr[0] =   0x00;
  heth.Init.MACAddr[1] =   0x80;
  heth.Init.MACAddr[2] =   0xE1;
  heth.Init.MACAddr[3] =   0x00;
  heth.Init.MACAddr[4] =   0x00;
  heth.Init.MACAddr[5] =   0x00;
...
}
```

But heth.Init.MACAddr is defined as a uint8_t * here:

https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_eth.h#L366-L382

This causes a 0-Pointer access, when accessing MACAddr[0].

ALABSTM, 2020-04-10:
Hi @thomask77,

To report issues related to the "Cube MX" tool or ask questions about, there is a dedicated [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) in the ST Community forum.

Regarding the point you reported, I would say both steps are supposed to be **handled by the user** within the `main.c` file:
* the array declaration in the "Private variables" dedicated section.
```c
/* USER CODE BEGIN PV */
uint8_t macaddr[6];
/* USER CODE END PV */
```
* the pointer initialization in the user code dedicated section.
```c
int main(void)
{
  /* USER CODE BEGIN 1 */
  heth.Init.MACAddr = macaddr;
  /* USER CODE END 1 */
  ...
}
```

In the fix you suggested, the array looks to be declared as **local** to the `MX_ETH_Init(void)`. Please check whether this might cause an issue at your application level during run-time.

I hope these few hints are helpful. Anyway, it would be preferable that you submit your "Cube MX" related questions to the the ST Community [link](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx). There you will find people better suited to answer such topics.

As issues should be related to the software published within this repository and not to ecosystem or hardware, please allow me to close this one. Thank you for your comprehension.

With regards,

closed 2020-04-10 by ALABSTM

thomask77, 2020-04-12:
Thanks for your answer. I'll post the issue to the CubeMX-Forum then, and post a link to this bug for the explanation.

> In the fix you suggested, the array looks to be declared as local to the MX_ETH_Init(void). Please 
> check whether this might cause an issue at your application level during run-time.

I've declared the array as `static`, so it's a global variable. Your solution is probably better aligned with the ST coding style though.

Perhaps a last request:

- Can you add an `assert_param` statement in `HAL_ETH_Init` and check the MAC pointer?
  That way, customers will find the problem when using `FULL_ASSERT`.

thomask77, 2020-04-12:
I referenced the issue here:
https://community.st.com/s/question/0D53W000004KWmMSAW/cubemx-0pointer-access-in-generated-mxethinit-code

Hope that's the right place.
Greetings, Thomas

thomask77, 2021-05-24:
@ALABSTM did you at least add an assert_param() statement like mentioned in https://github.com/STMicroelectronics/STM32CubeH7/issues/33#issuecomment-612609455 ?

Otherwise this issue will go unnoticed (as seen in the faulty auto-generated CubeMX code)

Please escalate!