STMicroelectronics/STM32CubeH7 #34: Dual Core boot up example

bytesontherocks, 2020-04-23:
**Describe the set-up**
 * STM32H757I-EVAL
 * STM32CubeIDE 1.3.1

**Additional context**
I'm following the templates for the Dual Core examples, when both cores boot at the same time. The CPU2 boots-up, gets into Stop mode. I can't see how the CPU gets out of Stop mode. I can see that CPU1 releases HSEM 0 but is this producing and event?

I can't make the code to work.

ALABSTM, 2020-04-29:
Hi @bytesontherocks,

Thank you for this report. It may be related to this issue described in pull-request #26. Within your [system_stm32h7xx_dualcore_boot_cm4_cm7.c](https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/CMSIS/Device/ST/STM32H7xx/Source/Templates/system_stm32h7xx_dualcore_boot_cm4_cm7.c) file, update the line indicated below:
```diff
  /* Configure the Vector Table location add offset address ------------------*/
#ifdef VECT_TAB_SRAM
-  SCB->VTOR = D2_AHBSRAM_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal SRAM */
+  SCB->VTOR = D2_AXISRAM_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal SRAM */
#else
  SCB->VTOR = FLASH_BANK2_BASE | VECT_TAB_OFFSET; /* Vector Table Relocation in Internal FLASH */
#endif
```
Would you mind trying this solution and reporting whether it solved the problem or not? Thank you.

With regards,

ALABSTM, 2020-05-04:
Hi @bytesontherocks,

Any feedback about this issue?

Thank you,

bytesontherocks, 2020-05-04:
Hi @ALABSTM, 
I haven't been able to test it yet but I was not using NVIC from RAM, so I don't think is this issue. I will re-run some of the test I did with some change in the eval board and I'll report. 
Thanks

ALABSTM, 2020-05-04:
Hi @bytesontherocks,

Thank you for your reply. I will wait for your updated report.

With regards,

bytesontherocks, 2020-05-05:
Hi @ALABSTM,
Imported a fresh template of BootCM4CM7 project from SW4STM32 to STM32CubeIDE (1.3.1). I set up the project the following way: 
- M7 debug session flashes both images 
- M4 debug session attaches to the system instead of start a debug session

To have some visual feedback I added for each core a different blinking LED.

M7 debug session started correctly but it goes to error_handler. The issue is mentioned here: https://visualgdb.com/tutorials/arm/stm32/multicore/startup/

so I replace the while loop with the timeout with while(__HAL_RCC_GET_FLAG(RCC_FLAG_D2CKRDY) != RESET) { asm("NOP");}

That didn't improve it, the M7 gets stuck waiting at RCC_FLAG_D2CKRDY, while the M4 can not attach (STLink doesn't connect) but starting a new debug session for the M4 without reseting the core connects but gets stuck in reset handler. 

To rule out that is not my debug settings what are causing this situation I programmed both cores with STM32CubeProgrammer, no LEDS flashing either.

I removed the HSEM and initial RCC_FLAG_D2CKRDY checks in the M7 and the HSEM and going to Stop from M4: Both LEDS blinking and I am able to debug both cores.

Both OB BM4 and BCM7 are set.

I think there are some issues with waking up from the Stop mode in the M4. 

Are you able to tell me if I'm doing something wrong or is the code that is not fully set up?

ALABSTM, 2020-05-06:
Hi @bytesontherocks,

Thank you for this detailed report and for the explanatory link you provided. Your report will be forwarded to our development teams to be analysed. I will be back to you as soon as I get their feedback.

But allow me question before: do you confirm you are using the latest firmware published (i.e. STM32CubeH7 **v1.7.0**)? Thank you.

With regards,

bytesontherocks, 2020-05-06:
Yes I do, I cloned the repo from this github location and I double checked is up to date v1.7.0.

ALABSTM, 2020-05-14:
Hi @bytesontherocks,

I hope you are doing well. In the [BootCM4_CM7](https://github.com/STMicroelectronics/STM32CubeH7/tree/master/Projects/STM32H747I-EVAL/Templates/BootCM4_CM7) example, there are two confusingly similar while() loops implemented from the CM7 side:
* First one waiting for the CM4 to **enter** the "stop" mode.
https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Projects/STM32H747I-EVAL/Templates/BootCM4_CM7/CM7/Src/main.c#L62
* Second one waiting for the CM4 to **wake up from** the "stop" mode.
https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Projects/STM32H747I-EVAL/Templates/BootCM4_CM7/CM7/Src/main.c#L95

According to you comment, you set the the `asm("nop");` instruction within the first loop. However, would it be possible that actually you **inadvertently** set it within the second one?

Please excuse the question. It is simply meant to remove this possibility from the list of potential causes of the issue you are facing.

Thank you,

bytesontherocks, 2020-05-14:
HI @ALABSTM,
I tried the following options for the loops you mentioned:
- with ErrorHandler() + timeout as per the example 
- with asm("nop"); in both while loops that you have indicated.

Neither worked, neither debugging or not.

Replying directly to your question, yes, I placed asm("nop"); to Line 95 but not inadvertently, I thought that was the right thing to do when debugging.

The implementation of the ErrorHandler() in the BootCM4_CM7 example is a while(1). The difference between executing asm("nop") you don't give up and keep checking the RCC_FLAG_D2CKRDY and with the ErrorHandler and timeout the CM7 core does nothing else, Is there some timing effect that I should be aware of?

Thanks.

ALABSTM, 2020-05-26:
Hi @bytesontherocks,

Please have a look to [this](https://www.st.com/resource/en/application_note/dm00629855-getting-started-with-projects-based-on-dualcore-stm32h7-microcontrollers-in-stm32cubeide-stmicroelectronics.pdf) application note available [here](https://www.st.com/en/development-tools/stm32cubeide.html#) (scroll down to "APPLICATION NOTES" section) on our st.com website.

Chapter 3 at page 12 explains how to debug dual core applications using anyone of the two debuggers (STLINK GDB server or OpenOCD) offered by Cube IDE.

I hope this will help you overcome the issue you are facing. Please let me know if this is the case or if you still need help.

With regards,

bytesontherocks, 2020-05-27:
Hi @ALABSTM, I followed that before asking. Also, as I reported previously, to make sure it was not due to debugging settings, I tried to execute the binaries flashing them with STCubeProgrammer, with same results.

I assume you (somebody in ST) has tested this example/template in the mentioned eval board, could you provide the Option Bytes values that you use to test it, please?

ALABSTM, 2020-06-12:
Hi @bytesontherocks,

According to our development teams, the project structure of multi-cpu examples has significantly evolved in  Cube IDE compared to SW4STM32. Below is a solution they suggest you the solution below hoping it will definitely solve the problem you are facing.

Regarding the options bits, in case you have not modified the default values there shall be no issue from that side. However, in case you always want to have the configuration used, do not hesitate.

The solution is to create a new project from scratch using the Cube IDE. Here are the steps to follow:
* Create a new H7 DC (Dual Core) project based on the HAL.
* In the main() function of the CM7 sub-project, modify the **first** while loop as you already did.
  ```diff
  -while((__HAL_RCC_GET_FLAG(RCC_FLAG_D2CKRDY) != RESET) && (timeout-- > 0));
  -if (timeout < 0) { Error_Handler(); }
  +while(__HAL_RCC_GET_FLAG(RCC_FLAG_D2CKRDY) != RESET) { asm("NOP"); }
  ```
* Build both CM7 and CM4 sub-projects.
* Create a **first** debug configuration for the **CM7** sub-project using the **ST-Link GDB Server** probe.
  * From the startup tab, load **both** binaries into the flash memory.
  * Screenshots:
    * Startup tab.
[image]
    * Debugger tab.
[image]
* Create a **second** debug configuration for the **CM4** sub-project using the **ST-Link GDB Server** probe.
  * From the startup tab, **do not** load CM4-related binary into the flash memory.
  * From the debugger tab, perform also an **Autostart** of the server, but pay **attention** to:
    * Increment the GDB port number (+2 for example).
    * Set the reset type to **none**.
  * Screenshots:
    * Startup tab.
[image]
    * Debugger tab.
[image]
* Run the CM7 core.
  * Code execution should break at the level of the first instruction, **i.e.** the modified while loop.
* Run the CM4 core.
  * Code execution should break at the level of the first instruction, **i.e.** `__HAL_RCC_HSEM_CLK_ENABLE();`.
  * Set a break point on the `HAL_Init();` instruction. This will serve later.
  * So far, this is how your Cube IDE window should look like.
[image]
* Resume the CM4 code execution.
  * It shall enter the sleep mode.
  * On the CM4 sub-project window, you should read that execution broke at address **0xa05f0000**.
[image]
* Resume the CM7 code execution.
  * The CM7 core will wake the CM4.
  * The GDB client is not aware that the CM4 core woke up. Nothing particular will be displayed on the CM4 sub-project window.
* Now, simply click on the "suspend/resume" button of the CM4 sub-project.
  * The window should display the C code again.
  * Code execution should halt at the level of the `HAL_Init();` instruction (thanks to the beak point we set earlier).
[image]
* Now you can debug execution on both cores as you wish.

I hope this tutorial will finally help you enjoy this dual core example. Please do not hesitate in case you still face some issue.

With regards,

bytesontherocks, 2020-06-24:
Hi, 
Thanks for your instructions.

I follow them:
-debug M7 and waits into the loop as expected
-launch debug or run M4 and the following error appears:

Error in final launch sequence:

Failed to start GDB server
Failed to start GDB server
Error in initializing ST-LINK device.
Reason: (18) Could not verify ST device! Abort connection.

This is a genuine H7 eval board, using STLink v3 onboard. Any clue where is the problem?

Thanks.

[image]

ALABSTM, 2020-07-22:
Hi @bytesontherocks,

Please excuse my delayed reply. I will forward this to our development teams hoping they will help you definitely solve the problem.

Once again, excuse this delayed reply and thank you for your patience.

With regards,

ALABSTM, 2020-07-29:
Hi @bytesontherocks,

I have feedback from our development teams:
* They asked whether you checked the "Halt all cores" option in the "Debugger" tab of the **CM7** sub-project (bottom of the screenshot below).
* In case you did, however, but still get the error message, they suggested the following test:
  * From the **CM7** debugging session:
    * Once you obtain the error message you mentioned...
    * ... resume code execution again to **wake up** the CM4 core.
  * From the **CM4** debugging session:
    * Reconnect to the **CM4** core (by simply restarting the debugging session for example).
    * Resume code execution.
    * Code execution should halt at the level of the `HAL_Init();` instruction (if a beak point is still set there).

We are looking forward to reading your feedback to know whether it is a simple foresight to activate the "Halt all cores" option or, in the contrary case, whether the **CM4** core execution resumed normally after it has been woken up by the **CM7** core.

With regards,

<p align="center">
  [image]
</p>

bytesontherocks, 2020-08-06:
hi @ALABSTM,

Thanks for your reply. After following your instructions I was having issues. Finally I started having a look at the Option Bytes (this is why some posts ago I asked about the OB settings but I suppose it was forgotten about). NRST_STOP_D2 and NRST_STBY_D2 were unchecked and that was preventing to M4 to resume, so I checked them both, even if NRST_STOP_D2 sould be the one to check for this particular issue.

I would recommend you to include the OB settings in your documentation on how to debug Dual Core.

I close the issue and thanks again.

[image]

closed 2020-08-06 by bytesontherocks

ALABSTM, 2020-08-07:
Hi @bytesontherocks,

Really glad to know you overcame this problem. Thank you for all the details you provided. I will let our development teams know about this OB-related aspect and about your suggestion. Thank you once more for your contribution.

With regards,

ash1729, 2021-03-29:
> hi @ALABSTM,
> 
> Thanks for your reply. After following your instructions I kept having issues. Finally I started having a look at the Option Bytes (this is why some posts ago I asked about the OB settings but I suppose it was forgotten about). NRST_STOP_D2 and NRST_STBY_D2 were unchecked and that was preventing to M4 to resume, so I checked them both, even if NRST_STOP_D2 sould be the one to check for this particular issue.
> 
> I would recommend you to include the OB settings in your documentation on how to debug Dual Core.
> 
> I close the issue and thanks again.
> 
> [image]

Hi, Where can I change the OB settings ? is it in the IDE

ash1729, 2021-03-29:
> Hi @bytesontherocks,
> 
> Really glad to know you overcame this problem. Thank you for all the details you provided. I will let our development teams know about this OB-related aspect and about your suggestion. Thank you once more for your contribution.
> 
> With regards,

Hi, where can I change this OB settings?