STMicroelectronics/STM32CubeH7 #37: Spelling: Lots of issues found by codespell

thomask77, 2020-05-01:
Here's the full output of `codespell -q4`:

https://gist.githubusercontent.com/thomask77/c65d7d1017e88ae0dd47d8b416fef0cb/raw/ae528003b68cccb1052ed98b0b48e5c0b638f991/spelling.txt

Some false positives, but mostly real errors.

You should be able to fix them semi-automatically, see  <https://github.com/codespell-project/codespell>

Best regards,
Thomas

RKOUSTM, 2020-05-14:
Hi @thomask77, 

Thank you for your contribution. Your report will be transmitted to our development teams to have this corrected in the frame of a future release. 

However, typos  reported within third_Party (FreeRTOS and LwIP for instance)  and CMSIS Core files are out of the scope of STMicroelectronics.  What about reporting to the concerned parties directly?

Thank you also for the link to the **codespell** project. It has been forwarded to our quality teams in order to study its use with our set of other tools. 

Thank you again for your contribution.

With regards,

RKOUSTM, 2020-05-18:
ST Internal Reference: 86525 86527 86529 86530 86531

RKOUSTM, 2020-05-18:
ST Internal Reference: 86527

RKOUSTM, 2020-05-18:
ST Internal Reference: 86529

RKOUSTM, 2020-05-18:
ST Internal Reference: 86530

RKOUSTM, 2020-05-18:
ST Internal Reference: 86531

RKOUSTM, 2021-03-11:
Hi @thomask77,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.9.0 of the STM32CubeH7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-11 by RKOUSTM