STMicroelectronics/STM32CubeH7 #38: FreeRTOS-MPU does not use the MPU on M7 cores

Palantir555, 2020-05-05:
I've reported this bug through your [forums](https://community.st.com/s/question/0D53W000003yAyDSAU/unable-to-set-up-noncacheable-mpu-regions-for-lwipethernet-when-using-freertosmpu-on-stm32h743zi) and the support portal. Feel free to close the issue if it's already been escalated/solved internally.

- Hardware: Nucleo-H743ZI2
- IDE: IAR for Arm v8.32.1

All M7 cores are affected, regardless of the compiler used.

## The bug

The FreeRTOS-MPU ports provided for M7 cores (ARM_CM4_MPU and ARM_CM7_MPU), and possibly others, do not actually use the MPU.

From `ARM_CM4_MPU/port.c`:

```
    #define portEXPECTED_MPU_TYPE_VALUE             ( 8UL << 8UL ) /* 8 regions, unified. */
```

M7 cores support 16 MPU regions, so that MPU->TYPE value is incorrect. As a result, the port silently fails to initialize the memory protection unit.

The fix:

```
    #if defined(CORE_CM4)
        #define portEXPECTED_MPU_TYPE_VALUE         ( 8UL << 8UL ) /* 8 regions, unified. */
    #else
        #define portEXPECTED_MPU_TYPE_VALUE         ( 16UL << 8UL ) /* 16 regions, unified. */
    #endif /* defined(CORE_CM4) */
```

That `#if defined(CORE_CM4)` is in line with the implementation at stm32h7xx_hal_cortex.h

## How to reproduce

1. Run the FreeRTOS-MPU example from the SDK on an M7 MCU
2. Use your debugger of choice to check the MPU->CTRL register. It will show that the MPU is disabled
3. Implement the fix mentioned above, and MPU->CTRL will show the MPU is now enabled, as expected

## After this fix

After fixing this bug, a different one came up. A crash within the assembly routine `vPortRestoreContextOfFirstTask` of the ARM_CM4_MPU FreeRTOS port. I'm still debugging that new one, and awaiting response from your support guys.

If you'd like any more info, let me know.

RKOUSTM, 2020-05-14:
Hi @Palantir555

Thank you for your contribution. Could you please confirm you are using the latest firmware version (STM32Cube H7 v1.7.0) and that the issue is systematically reproducible on your side?

With regards,

Palantir555, 2020-05-14:
Yeah, I'm at 1.7.0 (HEAD of `master`) at the moment.

The issue is 100% reproducible.

RKOUSTM, 2020-05-14:
Hi @Palantir555,

Thank you for having reported this point. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get feedback.

Thank you for your patience and thank you again for your contribution.

With regards,

Palantir555, 2020-05-25:
To add to this, other parts of the port rely on Cortex M4-specific MPU capabilities. e.g:

```
#define portLAST_CONFIGURABLE_REGION            ( 7UL )
```

expects the MPU to support up to 8 regions, while the Cortex M7 supports up to 16. IF you are planning expanding the port's awareness of those other 8 regions, please  remove the 2 pieces of code within `vPortStoreTaskMPUSettings` that wipe out any MPU region not allocated by FreeRTOS.

Or provide an alternative way to allocate regions taking full advantage of the MPU HAL's capabilities.

rahmanih, 2020-06-01:
Hi @Palantir555,
Sorry for the delay,
Thanks indeed for the time spent to analyse and report the issue.
The issue is indeed confirmed, we are working to fix it.

keep you updated.

regards
Haithem.

Palantir555, 2020-06-01:
Thanks for the update Haithem. Let me know if there's anything I can do to help.

BTW, in case you haven't seen it, there's more info on that "after this fix" note here: https://github.com/STMicroelectronics/STM32CubeH7/issues/39

RKOUSTM, 2020-06-01:
ST Internal Reference: 87368

RKOUSTM, 2021-06-23:
Hi @Palantir555,
 
I hope you are fine.

The issue you reported has been fixed in the frame of FreeRTOS Kernel V10.3.1. In addition, this version of FreeRTOS is integrated on the STM32CubeH7 V1.9.0 published on GitHub.

Thank you again for having reported.

With regards,

closed 2021-06-23 by RKOUSTM