STMicroelectronics/STM32CubeH7 #39: FreeRTOS-MPU fails to run if its privileged RAM is in a cacheable memory region

Palantir555, 2020-05-22:
## The setup

- Hardware: Nucleo-H743ZI2
- IDE: IAR for Arm v8.32.1

## The bug

FreeRTOS-MPU faults during initialization because of a cache coherency issue with `pxCurrentTCB`; allocated in the Privileged RAM region.

The SDK example only works because the Privileged RAM section is allocated in DTCM, a non-cacheable memory region.

To solve it, that memory region should be set as non-cacheable in the MPU. That way it can be allocated in cacheable memory regions.

In more technical detail:

- FreeRTOS-MPU allocates `pxCurrentTCB` in the privileged RAM section.
- During `xPortStartScheduler`, the privileged RAM section is allocated in the MPU as cacheable
- As a result, if the DCache is enabled, pxCurrentTCB will be cached.
- Cache coherency is lost during the initialization of the RTOS' scheduler, the assembly routine `vPortRestoreContextOfFirstTask` attempts to access a Reserved memory region, and results in a hard fault

## How To Reproduce

0. Grab the FreeRTOS-MPU sample application from the SDK
1. Fix the FreeRTOS-MPU bug described in issue #38 so the MPU is enabled
2. Update the linker script and `main.c` to allocate the Privileged Data section into AXI ram instead of DTCM
3. If you're using different hardware than me, make sure the DCache is enabled and has access to the AXI SRAM region you set up in the previous step
4. Try to run the application, and you should get a crash right away. Or perhaps some other serious misbehavior, depending on which incorrect values are loaded into your registers from `vPortRestoreContextOfFirstTask`

## Additional context

For a minimum viable fix, a couple of things need to be updated in each FreeRTOS-MPU port affected by this bug:

1. `portmacro.h` needs to support "normal, shareable, non-cacheable" configurations of MPU->RASR registers. That's `TEX=001 S=1 B=0 C=0` according to [the docs](https://developer.arm.com/docs/dui0646/a/cortex-m7-peripherals/optional-memory-protection-unit/mpu-access-permission-attributes#BIHHEDAE):
    `#define portMPU_REGION_SHAREABLE_NON_CACHEABLE ( 0x0cUL << 16UL )`
2. Privileged RAM needs to be configured using that `portMPU_REGION_SHAREABLE_NON_CACHEABLE` **instead of** `portMPU_REGION_CACHEABLE_BUFFERABLE`.
    - In `prvSetupMPU`:
    ```
         portMPU_REGION_ATTRIBUTE_REG =  ( portMPU_REGION_PRIVILEGED_READ_WRITE ) |
    #if 0 /* Original code */
                                         ( portMPU_REGION_CACHEABLE_BUFFERABLE ) |
    #else /* fix */
                                         ( portMPU_REGION_SHAREABLE_NON_CACHEABLE ) |
    #endif
                                         prvGetMPURegionSizeSetting( ( uint32_t ) __privileged_data_end__ - ( uint32_t ) __privileged_data_start__ ) |
                                         ( portMPU_REGION_ENABLE );
    ```
    - In `vPortStoreTaskMPUSettings`:
    ```
        xMPUSettings->xRegion[ 1 ].ulRegionAttribute =
                ( portMPU_REGION_PRIVILEGED_READ_WRITE ) |
    #if 0 /* Original code */
                ( portMPU_REGION_CACHEABLE_BUFFERABLE ) |
    #else /* fix */
                ( portMPU_REGION_SHAREABLE_NON_CACHEABLE ) |
    #endif
                prvGetMPURegionSizeSetting( ( uint32_t ) __privileged_data_end__ - ( uint32_t ) __privileged_data_start__ ) |
                ( portMPU_REGION_ENABLE );
    ```
Alternatively, memory barriers could be used to ensure the coherency of pxCurrentTCB, but I was not successful in my attempts, and I'm not sure if other FreeRTOS variables would still be affected in more insidious ways.

## More context

The fix described earlier got me past this problem, but I'm still running into other crashes shortly after. These may or may not have related causes; I'm still investigating.

If you uncover any other issues while fixing this one, a heads-up would be greatly appreciated.

Thanks,
JC

Palantir555, 2020-05-25:
Heads-up: I'm still finding this problem. The fix mentioned above does not seem to be working -it apparently did at first, so it must have been a problem in my debug setup. I'm still investigating the cause and solution, so any insight would be appreciated.

The description of the bug and replication steps remain the same.

RKOUSTM, 2020-05-27:
Hi @Palantir555,

Nice to hear from you again. Thank you for these details. 

You message will be forwarded to our development teams  for deeper analysis. We will be back to you as soon as we get an answer. Thank you for your patience and thank you, once more, for having reported the issue.

With regards,

RKOUSTM, 2020-06-01:
ST Internal Reference: 87368

Palantir555, 2020-06-15:
Hi there,

I understand these problems take time, but any chance you've made progress on this?
Do you have any hints on what the root of the problem is?

This issue is halting progress on a couple of our projects, and we can't wait until the next SDK release. I can fix the issue myself if you can help me find it.

Thanks for your time,
JC

rahmanih, 2020-06-15:
Hi @Palantir555,

Is it possible to share with us a "sample" project reproducing the issue?

regards
Haithem.

Palantir555, 2020-06-15:
Sure thing, appreciate the quick response.

I've reproduced the issue in [my fork (branch `reproduce_freertos_cache_problem`)](https://github.com/demux-labs/STM32CubeH7/tree/reproduce_freertos_cache_problem)

Please check the commit history to see the changes it required.

Most significant changes:

1. I had to move (copy) the `ARM_CM4_MPU` port folder to the SDK's root dir, because the relative path to `portmacro.h` was too long for IAR's compiler (🤦‍♂️)
2. Implemented the change described in #39
3. Copied the `FreeRTOS_MPU` example to `demux_FreeRTOS_MPU`, updated its linker config and main.c to move all data to AXI

rahmanih, 2020-06-17:
Issue reproduced!

looking at it, get back to you as soon as possible.

regards
Haithem.

rahmanih, 2020-07-02:
Hi @Palantir555,

root cause found and issue fixed. it is indeed a bug in FreeRTOS kernel.
to apply the fix, do the following:
in the <code> IAR/ARM_CM4_MPU/port.c </code> replace the code:
``
#define portMPU_REGION_CACHEABLE_BUFFERABLE						( 0x07UL << 16UL )
``
with
``
#define portMPU_REGION_CACHEABLE_BUFFERABLE						( 0x03UL << 16UL )
``
the original value sets the 'S' bit to 1 which disables the cache, this is the root cause of the issue.

could try and let us know?

regards
haithem.

Palantir555, 2020-07-03:
Hey Haithem, that's great news! Thanks a lot for your time and support.

I've run the example from my fork using your fix and it seems to be working correctly. When I run our more advanced codebase (LwIP included), it still crashes with a memory error, but it's different. It could be completely unrelated. Let me look further into that and report back.

BTW, you mentioned that enabling 'S' enabled the cache. Could you please elaborate on that?  If I'm understanding [this table](https://developer.arm.com/documentation/dui0646/a/cortex-m7-peripherals/optional-memory-protection-unit/mpu-access-permission-attributes#ftn.CIHIEJAJ) correctly, in that scenario 'S' only affects the shareability of the region; not its cacheability. I'm trying to figure out whether I misunderstood something, or if I'm looking at the wrong piece of documentation.

Thanks again for your effort! It's really appreciated.

Cheers,
JC

rahmanih, 2020-07-03:
the Cache L1 is per core, and can't be shared with other cores, thus enabling the 'S' bit leads to cache disablement.

regards
Haithem.

rahmanih, 2020-07-06:
Hi @RKOUSTM,

this is resolved now, we can close this ticket.

regards
Haithem.

ALABSTM, 2020-07-27:
Ticket reopened. Will be closed once the fix is published.

RKOUSTM, 2021-06-23:
Hi @Palantir555, 

I hope you are fine.

The issue you reported has been fixed in the frame of FreeRTOS Kernel V10.3.1. In addition, this version of FreeRTOS is integrated on the STM32CubeH7 V1.9.0 published on GitHub.

Thank you again for having reported.

With regards,

closed 2021-06-23 by RKOUSTM