STMicroelectronics/STM32CubeH7 #40: Missing CMakeLists

WBABD, 2020-05-26:
Please can you add a CMakeLists for this project

ALABSTM, 2020-06-08:
Hi @WBABD,

Thank you for having tackled this aspect. You request has been forwarded to our development teams. We will be back to you as soon as they provide us with their answers.

With regards,

ALABSTM, 2021-06-17:
Hi @WBABD,

I hope you are doing well. Excuse this late answer. Unfortunately, we do not provide `CMakeLists.txt` files within our firmware packages. However, you can find several initiatives from **STM32** users providing such files. You can, for instance, check the [stm32-cmake](https://github.com/ObKo/stm32-cmake) project.

I hope this helps. Please allow me to close this issue now.

With regards,

closed 2021-06-17 by ALABSTM