STMicroelectronics/STM32CubeH7 #41: Uninitialized local variable VidCfg

cornway, 2020-06-01:
https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Drivers/BSP/STM32H747I-DISCO/stm32h747i_discovery_lcd.c#L703

Sometimes might cause undefined behaviour of DSI host controller (incorrect configuration),
especially might happen when **stack** area is situated out of ZI-section.

RKOUSTM, 2020-06-17:
Hi @cornway,

Thank you for reporting this issue. 

In order to allow a better understanding of the problem, could you please describe how you did to detect this issue and how to reproduce it, so that our development teams can give you an answer.

With regards,

cornway, 2020-06-18:
Hi @RKOUSTM , that issue happened during custom project developing.
Project is totally based on Stm32 Cube layers, for my case STR was as following :

> Keil MDK IDE, using custom scatter file (used by linker), -O0 -Otime, everything else is default
> Mark STACK area as UNINIT in scatter file, any ram region can be used
> Build project

In case when produced image size is lower than 130kbytes - no issue present,
otherwise - DSI is configured to fill frame buffer with colour bar, due garbage in  VidCfg variable (in uninitialized fields exactly).

So i don't have a STR based on template projects provided with H7 Cube library, but i suppose -
such issue can occur when at least one condition met - STACK should be uninitialized.

Note: for my case the fix was simple - just do `DSI_VidCfgTypeDef VidCfg = {0};`

Thank, best regards.

RKOUSTM, 2020-06-22:
Hi @cornway ,

Thank you for your contribution. You request will be forwarded to the development team in charge of BSP driver. I will let you updated once they provide me with their answer. Thank you for your patience and thank you for your message once again.

With regards,

RKOUSTM, 2020-07-02:
Hi @cornway,

The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

RKOUSTM, 2020-07-02:
ST Internal Reference: 89424

RKOUSTM, 2020-07-06:
Hi @cornway,

Thank you for your suggestion and contribution.  According to our development teams, the issue should be related to the non-initialization of the **FrameBTAAcknowledgeEnable** parammeter in **VidCfg** structure which is declared at the following BSP weak function: 

- ___weak HAL_StatusTypeDef MX_DSIHOST_DSI_Init(DSI_HandleTypeDef *hdsi, uint32_t Width, uint32_t Height, uint32_t PixelFormat)._

As consequence this field can take any value and can lead to the inspect behavior. The suggested solution that will be published as soon in the future release looks like the following:
 
[image]

Thank you again for your contribution.

With regards,

cornway, 2020-07-07:
You're welcome!
Regards.

RKOUSTM, 2021-03-11:
Hi @cornway,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.9.0 of the STM32CubeH7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-11 by RKOUSTM