STMicroelectronics/STM32CubeH7 #42: Missing include in etc.c

rolandvs, 2020-06-04:
How to:
- STM32CubeMX used to generate NUCLEO-H743ZI basic setup with Ethernet
- Generate code for Makefile project using `make clean all`

Result:
- Compiling the generated project will generate warnings concerning `memset` function

Solution:
- Add <string.h> to the `etc.c` file

ALABSTM, 2020-07-21:
Hi @rolandvs,

Thank you for this report. We could find no file named `eth.c`. May I ask you the complete name of the file and where exactly the call to `memset()` could be found?

Thank you,

ALABSTM, 2020-08-03:
Hi @rolandvs,

Any feedback?

Thank you,

ALABSTM, 2020-08-26:
Hi @rolandvs,

I hope you are fine. As there has been no feedback since more than a month, please allow me to close this issue. You can reopen it in case you think it is still relevant.

With regards,

rolandvs, 2020-09-09:
Dear @ALABSTM ,
I was preoccupied in some projects, but  just generated a project to show you the "problem" I reported earlier.

I have included the  [GENERIC.ioc.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/5197718/GENERIC.ioc.zip)

It is a `makefile` project and compiling it reveals the missing `string.h`...

```
Core/Src/eth.c: In function 'MX_ETH_Init':
Core/Src/eth.c:79:3: warning: implicit declaration of function 'memset' [-Wimplicit-function-declaration]
   79 |   memset(&TxConfig, 0 , sizeof(ETH_TxPacketConfig));
      |   ^~~~~~
Core/Src/eth.c:79:3: warning: incompatible implicit declaration of built-in function 'memset'
Core/Src/eth.c:22:1: note: include '<string.h>' or provide a declaration of 'memset'
   21 | #include "eth.h"
  +++ |+#include <string.h>
   22 |
```

ALABSTM, 2020-09-16:
Dear @rolandvs,

Good to read your message. Thank you for these extra details. The issue is reopened and will be forwarded to our development teams. I will get back to you as soon as they provide me with feedback.

With regards,

ALABSTM, 2020-09-28:
Hi @rolandvs,

Our development teams confirmed what you pointed out. This is a Cube MX-related issue. It will be fixed and made available in a future Cube MX release.

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to our ecosystem (namely the Cube MX tool), please allow me to close it.

Thank you again for your contribution and thank you for your patience. We are looking forward to reading from you again.

With regards,

ALABSTM, 2020-09-28:
ST Internal Reference: 82323

closed 2020-09-28 by ALABSTM