STMicroelectronics/STM32CubeH7 #46: HAL_RCCEx_PeriphCLKConfig not setting the HRTIM1 clock source

andigandi, 2020-06-30:
STM32 Nucleo-144 for STM32H743
Cube IDE 1.3.0 with STM32CubeH7 1.7.0

Description:
Using `HAL_RCCEx_PeriphCLKConfig` with `PeriphClkInitStruct.Hrtim1ClockSelection = RCC_HRTIM1CLK_CPUCLK;` with the intent of setting the clock source for HRTIM1 peripheral to the CPU clock has no effect.

The problem seems to originate in stm32h7xx_hal_rcc_ex.c in line 1474. The function tries to check if the HRTIM1 peripheral does exist but it uses the wrong non-existing define `HRTIM` for that purpose:
```
#if defined(HRTIM)
  /*------------------------------ HRTIM1 clock Configuration ----------------*/
  if(((PeriphClkInit->PeriphClockSelection) & RCC_PERIPHCLK_HRTIM1) == RCC_PERIPHCLK_HRTIM1)
  {
    /* Check the parameters */
    assert_param(IS_RCC_HRTIM1CLKSOURCE(PeriphClkInit->Hrtim1ClockSelection));

    /* Configure the HRTIM1 clock source */
    __HAL_RCC_HRTIM1_CONFIG(PeriphClkInit->Hrtim1ClockSelection);
  }
#endif  /*HRTIM*/
```

The file stm32h743xx.h defines `HRTIM1` in line 2393. Changing the above code to use this define instead fixes the problem. The right clock source gets applied by `HAL_RCCEx_PeriphCLKConfig`.
```
#if defined(HRTIM1)
  /*------------------------------ HRTIM1 clock Configuration ----------------*/
  if(((PeriphClkInit->PeriphClockSelection) & RCC_PERIPHCLK_HRTIM1) == RCC_PERIPHCLK_HRTIM1)
  {
    /* Check the parameters */
    assert_param(IS_RCC_HRTIM1CLKSOURCE(PeriphClkInit->Hrtim1ClockSelection));

    /* Configure the HRTIM1 clock source */
    __HAL_RCC_HRTIM1_CONFIG(PeriphClkInit->Hrtim1ClockSelection);
  }
#endif  /*HRTIM1*/
```

ALABSTM, 2020-07-01:
Hi @andigandi,

Thank you for this detailed report. It will be forwarded to our development teams. I will get back to you as soon as I get their answer.

With regards,

ALABSTM, 2020-08-06:
ST Internal Reference: 81500

ALABSTM, 2020-08-06:
Hi @andigandi,

Thank you for your report. The issue you reported has been already fixed and the fix is already available in the frame of version **v1.8.0** of the STM32CubeH7 firmware package.

With regards,

closed 2020-08-06 by ALABSTM