STMicroelectronics/STM32CubeH7 #47: ExtMem_Boot: hardfault caused by unaligned memory access

Paullake85, 2020-07-06:
Hello!

I'm playing with the STM32h7B3I-DK with the ExtMem_Boot and the LedToggling application. 
I'm trying to load bootload the both DATA and CODE in the internal ram as follows:

#define DATA_AREA           USE_INTERNAL_SRAM  	//    //
#define CODE_AREA          USE_INTERNAL_SRAM		 // USE_EXTERNAL_SDRAM  //
#define BINARY_AREA        USE_SPI_NOR         // USE_SDMMC //

But after flashing both the boot and app binaries, the extboot application rise an hardfault caused by unaligned memory access when accesing the OSPI. While if I return in the original configuration with #define CODE_AREA  USE_EXTERNAL_SDRAM, everythings goes fine.

I do not know if it is caused by and error in the bootloader app or in the linking of the LedToggling app, heres how I updated the ld file of the LedToggling:

MEMORY
{
  ITCMRAM    (xrw)    : ORIGIN = 0x00000000,   LENGTH = 64K
  RAM    (xrw)    : ORIGIN = 0x24000000,   LENGTH = 1000K
  ROM    (rx)    : ORIGIN = 0xD0000000,   LENGTH = 8192K
}

/* Sections */
SECTIONS
{
  /* The startup code into "ROM" Rom type memory */
  .isr_vector :
  {
    . = ALIGN(4);
    KEEP(*(.isr_vector)) /* Startup code */
    . = ALIGN(4);
  } >RAM /*ROM*/

  /* The program code and other data into "ROM" Rom type memory */
  .text :
  {
    . = ALIGN(4);
    *(.text)           /* .text sections (code) */
    *(.text*)          /* .text* sections (code) */
    *(.glue_7)         /* glue arm to thumb code */
    *(.glue_7t)        /* glue thumb to arm code */
    *(.eh_frame)

    KEEP (*(.init))
    KEEP (*(.fini))

    . = ALIGN(4);
    _etext = .;        /* define a global symbols at end of code */
  } >RAM /*ROM*/

  /* Constant data into "ROM" Rom type memory */
  .rodata :
  {
    . = ALIGN(4);
    *(.rodata)         /* .rodata sections (constants, strings, etc.) */
    *(.rodata*)        /* .rodata* sections (constants, strings, etc.) */
    . = ALIGN(4);
  } >RAM /*ROM*/

  .ARM.extab   : { 
    . = ALIGN(4);
    *(.ARM.extab* .gnu.linkonce.armextab.*)
    . = ALIGN(4);
  } >RAM /*ROM*/
  
  .ARM : {
    . = ALIGN(4);
    __exidx_start = .;
    *(.ARM.exidx*)
    __exidx_end = .;
    . = ALIGN(4);
  } >RAM /*ROM*/

  .preinit_array     :
  {
    . = ALIGN(4);
    PROVIDE_HIDDEN (__preinit_array_start = .);
    KEEP (*(.preinit_array*))
    PROVIDE_HIDDEN (__preinit_array_end = .);
    . = ALIGN(4);
  } >RAM /*ROM*/
  
  .init_array :
  {
    . = ALIGN(4);
    PROVIDE_HIDDEN (__init_array_start = .);
    KEEP (*(SORT(.init_array.*)))
    KEEP (*(.init_array*))
    PROVIDE_HIDDEN (__init_array_end = .);
    . = ALIGN(4);
  } >RAM /*ROM*/
  
  .fini_array :
  {
    . = ALIGN(4);
    PROVIDE_HIDDEN (__fini_array_start = .);
    KEEP (*(SORT(.fini_array.*)))
    KEEP (*(.fini_array*))
    PROVIDE_HIDDEN (__fini_array_end = .);
    . = ALIGN(4);
  } >RAM /*ROM*/

  /* Used by the startup to initialize data */
  _sidata = LOADADDR(.data);

  /* Initialized data sections into "RAM" Ram type memory */
  .data : 
  {
    . = ALIGN(4);
    _sdata = .;        /* create a global symbol at data start */
    *(.data)           /* .data sections */
    *(.data*)          /* .data* sections */

    . = ALIGN(4);
    _edata = .;        /* define a global symbol at data end */
    
  } >RAM /*AT> ROM*/

Can you help me?

Thx,
Paul

ASELSTM, 2020-07-13:
Hi @Paullake85,

Thank you for reporting this issue. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get an answer. Thank you in advance for your patience.

With regards,

Paullake85, 2020-07-14:
Hello!

I recently solved the problem because for make it working you must change:

SCB->VTOR = APPLICATION_ADDRESS | VECT_TAB_OFFSET; /* Vector Table Relocation to APPLICATION_ADDRESS in preprocessor defines */

in the system_stm32h7xx.c

Can you explain me a way to do this realocation from the bootloader?
Moreover I do not understand the value BINARY_SIZE 33000U, How much byte are? and how to set it?

Thx
Paul

ASELSTM, 2020-07-23:
Hi @Paullake85, 

According to our development team, the configuration you have used
```
#define DATA_AREA USE_INTERNAL_SRAM
#define CODE_AREA USE_INTERNAL_SRAM
```
is not supported in the application. The available configurations are detailed in the readme file in the [Table 1. IAP implementation on STM32H7B3I-DISCO](https://github.com/STMicroelectronics/STM32CubeH7/blob/79196b09acfb720589f58e93ccf956401b18a191/Projects/STM32H7B3I-DK/Applications/ExtMem_CodeExecution/ExtMem_Boot/readme.txt#L123).

In order to be able to run the application with the chosen configuration ,which is the use of the internal SRAM for both Code and Data area,  you have to perform some modification as following:

1. Identify different address for each zone for example :

```
#define INTERNAL_SRAM1  0x24010000UL  for data  
#define INTERNAL_SRAM   0x24000000UL  for code   
```

2.  Insert the APPLICATION_ADDRESS to preprocessor defines and set an appropriate value 

`APPLICATION_ADDRESS=0x24000000U`

3. Copy the project.bin to OSPI memory and follow the same steps as BOOTROM example

Finally, to answer your question, the `BINARY_SIZE` correspond to the size of the generated project.bin of the application to be executed.

Best regards,

Paullake85, 2020-08-18:
Hello,
Thx for your replay. 
As I mentioned before the issue seems solved when I just modify:

SCB->VTOR = APPLICATION_ADDRESS | VECT_TAB_OFFSET; /* Vector Table Relocation to APPLICATION_ADDRESS in preprocessor defines */
in the system_stm32h7xx.c

What do you mean that I have define:
#define INTERNAL_SRAM1  0x24010000UL  for data  
#define INTERNAL_SRAM   0x24000000UL  for code

Do you mean that I have to define the new data address in the LD file for the linker? or should I modify the application system_stm32h7xx.c or the bootloader?

Finally, I would like to apply this bootloader to bigger binary size, unfortunately the resulting binary of the new application is too big to be flashed at 0x90000000, even after using the NOLOAD, is it possible to force the programmer to flash the HEX format at 0x90000000 (the hex is in interal ram space)

Thank you,
Paolo

ASELSTM, 2020-08-24:
Hi @Paullake85,

First, the dedicated address for the data and the code should be different if you use the same internal SRAM (0x24000000) for both data and code execution and it should be added within the `memory.h` file.

Second , you can change binary size manually within `memory.h` file or use the **STM32CubeProgrammer** to load the binary within the OSPI memory (0x90000000) with the dedicated memory flash loader mentioned in the readme file of the application.

Hope this would help you.
With regards,

Paullake85, 2020-08-26:
Hi!

My problem is that the binary size of the application is too big to be loaded in the flash about 1.6GB.

Thank you for your support.

Paul

ASELSTM, 2020-08-27:
Hi @Paullake85,

Unfortunately,  there's no possibility to force the programmer to flash the HEX  if the binary size of the application is too big to be loaded in the flash. You have to use another memory with greater size to load the entire code.

With regards,

ASELSTM, 2020-09-24:
Hi @Paullake85,

Please allow me to close this thread as no activity has been registered since a month. You can reopen it at any time if you need other information. Thank you for your comprehension.

With regards,

closed 2020-09-24 by ASELSTM