STMicroelectronics/STM32CubeH7 #51: Version control management

NGXTech, 2020-07-25:
**Caution**
Version had not been managed in the proper way.
**Describe the set-up**
Version 1.7.0 is not backward compatible with 1.5.0. So as per version control system, major number should be increased in that scenario.

**Additional context**
If you have a first analysis or a patch proposal, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2020-07-28:
Hi @NGXTech,

Thank you for your comment. May I ask you more details about the broken backward compatibility you mentioned? Would you mind list one or two examples of major changes you found causing compatibility break?

Thank you,

NGXTech, 2020-07-28:
> Hi @NGXTech,
> 
> Thank you for your comment. May I ask you more details about the broken backward compatibility you mentioned? Would you mind list one or two examples of major changes you found causing compatibility break?
> 
> Thank you,

Hi,

In stm32h750xx.h, GPIO_TypeDef structure earlier version, there were two members each uint16_t named BSRRH & BSRRL but in latest version, there are only 1 register named BSRR. This is a breaking the compatibilty.

ALABSTM, 2020-08-27:
Hi @NGXTech,

We compared the `stm32h750xx.h` file published in both **v1.5.0** and **v1.7.0** of the **STM32CubeH7** package but found no difference in the definition of the `BSRR` register.

Moreover, the `stm32h750xx.h` file has been **first** published in the frame of version **v1.5.0** of the firmware package and ever since, the definition of the `BSRR` register is the same (see below).
```c
  __IO uint32_t BSRR;     /*!< GPIO port bit set/reset,               Address offset: 0x18      */
```
Would you mind checking again from you side?

Thank you,

ALABSTM, 2020-09-07:
Hi @NGXTech,

Any feedback?

Thank you,

NGXTech, 2020-09-07:
> Hi @NGXTech,
> 
> Any feedback?
> 
> Thank you,

Hi,

I have checked it, sorry for the mistake, if you check the CubeFW version 1.3.0, in that GPIO structure is defined as BSRRH and BSRRL.

ALABSTM, 2020-09-17:
Hi @NGXTech,

May I ask you whether you experienced any issue at compilation- or run-time due to this update?

Examining the change I see it is **enclosed within** the body of the `` function without any impact on its prototype as shown below. Such a change is **transparent** at user application-level and shall yield **no compatibility break**.

```diff
void HAL_GPIO_WritePin(GPIO_TypeDef* GPIOx, uint16_t GPIO_Pin, GPIO_PinState PinState)
{
  /* Check the parameters */
  assert_param(IS_GPIO_PIN(GPIO_Pin));
  assert_param(IS_GPIO_PIN_ACTION(PinState));

  if(PinState != GPIO_PIN_RESET)
  {
+    GPIOx->BSRR = GPIO_Pin;
-    GPIOx->BSRRL = GPIO_Pin;
  }
  else
  {
+    GPIOx->BSRR = (uint32_t)GPIO_Pin << GPIO_NUMBER;
-    GPIOx->BSRRH = GPIO_Pin ;
  }
}
```

With regards,

ALABSTM, 2020-10-12:
Hi @NGXTech,

Any feedback?

With regards,

ALABSTM, 2020-10-13:
Hi @NGXTech,

Please allow me to close this issue. You can reopen it if you have anything else relevant to add regarding this topic.

With regards,

closed 2020-10-13 by ALABSTM