STMicroelectronics/STM32CubeH7 #54: Unable to import CDC Device Examples for CubeH7.

duaneellissd, 2020-08-29:
Step 1 - Install STM32 Cube IDE

Step 2 - Install (this package) for the STM32H7, installed via the IDE - it downloads the software package.

Step 3 - Attempt to install ANY USB Device Project, there are two, named "CDC_Standalone" - 

SELECT: Import STM32 Cube Example

Under MCU, select STM32H7
Under Board, select Evaluation board
Under middleware, select: USB_Device

Select project name: CDC_Standalone,
Location states:  Projects/STM32H743I-EVAL/Applications/USB_Device/CDC_Standalone
Description states: This application is part of the USB Device Library Package

Step 4: Click NEXT... project imports...

EXPECTED RESULT:  Source code would speak about "USB Device"

ACTUAL RESULT:  The HOST application is imported.

Step 5: Examine the project, you find that you have the HOST version instead.

NOTE: I do not think it matters, but I am running this on a MAC using the MAC version of tools.

duaneellissd, 2020-09-01:
NOTE: I think - I am not sure - that the problem is due to the HOST and DEVICE examples having the same name. These names should instead be unique.

ALABSTM, 2020-09-07:
Hi @duaneellissd,

Thank you for this report. However, this is rather a Cube MX related issue and I am afraid we do not address it here. Please submit it to the **ST Community**, in the dedicated [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx). There you shall find people who will address your request and whom you will hopefully find a solution with.

Please allow me to close this issue now. Thank you for your comprehension.

With regards,

closed 2020-09-07 by ALABSTM