STMicroelectronics/STM32CubeH7 #55: OB_USER_SWAP_BANK Undeclared on H750 Target

PGIII, 2020-08-31:
**Describe the set-up**
 Custom STM32H750 Board
arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)

**Describe the bug**
OB_USER_SWAP_BANK is Undeclared in expansion of OB_USER_ALL, OB_USER_SWAP_BANK is only declared when DUAL_BANK is defined and since the H750 does not declare DUAL_CORE, FLASH_OPTSR_VDDMMC_HSLV, or FLASH_OPTSR2_TCM_AXI_SHARED the H750 uses the default Declaration of OB_USER_ALL which includes OB_USER_SWAP_BANK 
see https://github.com/STMicroelectronics/STM32CubeH7/blob/master/Drivers/STM32H7xx_HAL_Driver/Inc/stm32h7xx_hal_flash_ex.h#L656

RKOUSTM, 2020-09-03:
Hi @PGIII,

Thank you for having reported this point. This issue is confirmed by our development teams. A fix will be implemented and published in the frame of a future release. Thank you once more.

With regards,

RKOUSTM, 2020-09-03:
ST Internal Reference: 92119

RKOUSTM, 2021-03-11:
Hi  @PGIII,
 
I hope you are fine. The issue you reported has been fixed in the frame of version v1.9.0 of the STM32CubeH7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-11 by RKOUSTM