STMicroelectronics/STM32CubeH7 #57: CMSIS_OS.C

rxa1031, 2020-09-15:
Hello,

C:\STM32CubeMX\Repository\STM32Cube_FW_H7_V1.8.0\Middlewares\Third_Party\FreeRTOS\Source\CMSIS_RTOS\CMSIS_OS.C

Line number 214
Code:
```
     if (xTaskCreate((TaskFunction_t)thread_def->pthread,(const portCHAR *)thread_def->name,
              thread_def->stacksize, argument, makeFreeRtosPriority(thread_def->tpriority),
              &handle) != pdPASS)  {
```

_**warning C4244**: 'function': conversion from 'const uint32_t' to 'const uint16_t', possible loss of data_

Other similar reporting:
[Similar issue link](https://stackoverflow.com/questions/59161343/has-cmsis-os-c-of-stm32cubemx-to-be-fixed)

Regards,
@rxa1031

RKOUSTM, 2020-09-17:
Hi @rxa1031,

First of all we would like to thank you for your report. In order to allow a better understanding of the problem, could you please give us more details allowing our development teams to reproduce the warning? 

IIn fact, for the first issue related to use correct ``TimerCallbackFunction_t`` type for timer creation  mentioned in the [link](https://stackoverflow.com/questions/59161343/has-cmsis-os-c-of-stm32cubemx-to-be-fixed), it is already fixed: https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.c#L377-L390

Thank you for your contribution.

With regards,

rxa1031, 2020-09-18:
Hi @RKOUSTM ,

May you please help me with the version which has the fix for `TimerCallbackFunction_t` are these files a part of the code generated by latest STM32CubeMX?

I use STM32CubeMX to generate initialization and FreeRTOS base code for STM32H753BIT6. 

The **disadvantages** being that:
1. The STM32CubeMX imports the FreeRTOS version 10.2.1 with CMSIS-RTOS version 2.00 (and not FreeRTOS version 10.4.0)
2. FreeRTOS Kernel cannot be compiled using Keil Compiler version 6. (I did not get help on this from any source).
3. STM32CubeMX generated code uses the functions from **CMSIS_OS.C file** (which are rappers over the FreeRTOS functions) for creating tasks.

**Current scenario:**
I am trying to test a piece of legacy 16 bit RTXC code by porting it to STM32H7 and FreeRTOS. I found that I need more break points than that currently supported by Keil, so I switched to FreeRTOS 10.3.1 simulator (FreeRTOS\DEMO\Win32-MSVC\WIN32.sln).
I imported the CMSIS_OS.C and CMSIS_OS.H files (that I retrieved from one of the STM32 FreeRTOS examples that is generated by STM32CubeMX) to the FreeRTOS simulator project / code.

After compilation, I am observing the shared concern / _warning_ for `xTaskCreate()`
The third argument is of type _configSTACK_DEPTH_TYPE_ which is typedef as below:
	`#define configSTACK_DEPTH_TYPE uint16_t`

The _reason for the issue_ is that `thread_def->stacksize` is of type `uint32_t`, which is the right type when passing to `xTaskCreateStatic`, but is not the write type when passing to `xTaskCreate`. Either the two FreeRTOS functions need to update their API argument type (in which case it may be better to update the xTaskCreate argument type) (_For this a request must be raised from your side to FreeRTOS_) or otherwise typecase the variable while passing it to argument 3 of xTaskCreate.

I am use the latest release FreeRTOS version 10.4.0, unless someone pushes STMicroelectronics to immediately upgrade their STM32CubeMX to allow using FreeRTOS version 10.4.0. I have already been asking them to move to FreeRTOS version 10.3.1 for quite a long time now. The biggest issue with Replacing /updating the FreeRTOS files is that the moment someone opens the IOC (project file) in STM32CubeMX and regenerates the code, the FreeRTOS 10.4.0 files will again get overwritten with FreeRTOS 10.2.1 files.

Regards,
@rxa1031

RKOUSTM, 2020-09-21:
Hi @rxa1031,

I hope you are doing well. First of all, the ``version V6.0.0`` of the STM32CubeMX  supports the fix of ``TimerCallbackFunction_t`` type for timer creation which is used in the ``CMSIS_OS.c`` file. Secondly, concerning the issue related to the compilation error using Keil when switching to compiler generation 6, you should update the lines below:
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.c#L62
 
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/CMSIS_RTOS/cmsis_os.c#L73

As follows:

```diff
-#if   defined ( __CC_ARM )
+#if defined ( __CC_ARM ) || (defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))
```
 ```diff
-#if   defined ( __GNUC__ )
+#if defined (GNUC) && !defined (__CC_ARM) && !(defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))
```
Finally, to answer your question, the ``configSTACK_DEPTH_TYPE`` correspond to third argument  for xTaskCreate() to be defined in the ``FreeRTOSConfig.h`` file of your application. This update is already noted in the comment below:
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/include/FreeRTOS.h#L839-L843

Thank you for your contribution.

With regards,

rxa1031, 2020-09-21:
Hi,

Does the below line of code has a typo?

#if defined (GNUC) && !defined (__CC_ARM) && !(defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))

should it be:
#if defined (__GNUC__) && !defined (__CC_ARM) && !(defined(__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050))

[FreeRTOS_ThreadCreation.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/5256322/FreeRTOS_ThreadCreation.zip)
Attached code (for STM32H743I-EVAL board) does not compile when using Arm / Keil Compiler version 6 even after making above changes.

I tied with both c99 and gnu99 as the informed C Language under Options for Target -> C/C++(AC6)

Regards,
Rajeev

rxa1031, 2020-09-21:
@RKOUSTM 
Please check below link:
https://github.com/FreeRTOS/FreeRTOS-Kernel/issues/154

and my previous message.

Regards,
@rxa1031

RKOUSTM, 2020-09-22:
Hi @rxa1031,

Thank you for your interest. We are presently working on this. A fix will be implemented and made available in future release. We will keep you informed.

With regards

RKOUSTM, 2020-09-22:
ST Internal Reference: 93121

rxa1031, 2021-03-23:
Hello @ALABSTM ,

Are there any findings / updates that you may be able to share with me.

Thanks,
Rajeev

RKOUSTM, 2021-06-24:
Hi @rxa1031,

In order to support the ARM V6 for the FreeRTOS application, we need to use the GCC port files for Keil Compiler Version 6 as follows: 

- Including this path ../../../../../../Middlewares/Third_Party/FreeRTOS/Source/portable/GCC/ARM_CM4F in the options for target of your project.

- Updating the port.c file to use the GCC corresponding definitions that are required for Keil Compiler Version 6. This update is available on the FreeRTOS Kernel [V10.4.0](https://www.freertos.org/2020/09/freertos-kernel-v10-4-0-is-now-available.html) version.

Thank you again for your contribution. Regarding the point for which you opened the issue, as it is resolved, please allow me to close the issue.

With regards,
Rihab

closed 2021-06-24 by RKOUSTM