STMicroelectronics/STM32CubeH7 #59: Issue with chosen MAX value (i.e. 56) of configMAX_PRIORITIES

rxa1031, 2020-09-21:
Hello,

I was reviewing the FreeRTOS code and the `CMSIS_OS2.C` file and the macro configMAX_PRIORITIES

I assume the below statement is incorrect and I have a reason for it:

```
/*  CMSIS-RTOSv2 defines 56 levels of priorities. To be able to use them
 *  all and avoid application misbehavior, configUSE_PORT_OPTIMISED_TASK_SELECTION
 *  must be set to 0 and configMAX_PRIORITIES to 56
 *
 */
```

Even the declarations like below are incorrect:
`#define configMAX_PRIORITIES                     ( 56 )`

**Observation 1:**
The FreeRTOS code version 10.3.1 (and maybe earlier) have a file named `blocktim.c` at location `FreeRTOSv10.3.1\FreeRTOS\Demo\Common\Minimal\blocktim.c`
The file declares the macro:
```
#ifndef bktSECONDARY_PRIORITY
	#define bktSECONDARY_PRIORITY	( configMAX_PRIORITIES - 4 )
#endif
```

I am not sure whether STM32CubeMX is aware of the above file or whether it has a different file in which above macro is created.

**Observation 2:**
File `CMSIS_OS2.C` declares the `enum named osPriority_t`, with the last three enumerated values as:
```
  osPriorityISR           = 56,         ///< Reserved for ISR deferred thread.
  osPriorityError         = -1,         ///< System cannot determine priority or illegal priority.
  osPriorityReserved      = 0x7FFFFFFF  ///< Prevents enum down-size compiler optimization.

```

**Conclusion:**

In order to allow code being ported to other microcontrollers (maybe non ST Microcontrollers) or debugged using MSVC, changes must be made to code.

From observation 1, `I believe all user implementations should only be allowed to use MAXIMUM PRIORITY value which is one less than bktSECONDARY_PRIORITY i.e.  configMAX_PRIORITIES - 5`

From observation 2 and above conclusion, I believe `the value assigned to configMAX_PRIORITIES should be osPriorityISR + 5`

Please share your opinion.

Regards,
@rxa1031

RKOUSTM, 2020-09-23:
Hi @rxa1031,

Thank you for this report. Unfortunately, for first point, we don't treat aspect related to FreeRTOS in our GitHub repositories. As the ``blocktim.c`` file is out of the scope of STMicroelectronics what about reporting this point to the [FreeRTOS GitHub](https://github.com/freertos)? Concerning the second point, the ``configMAX_PRIORITIES`` is set to 56 to support the ``osPriority`` in ``CMSIS V2``  (see [this](https://www.keil.com/pack/doc/CMSIS_Dev/RTOS2/html/group__CMSIS__RTOS__ThreadMgmt.html) link).

Please allow me to close this issue now. Thank you for your contribution. 

With regards,

rxa1031, 2020-09-23:
@RKOUSTM 

Thanks for sharing the response.

I understand the point that you shared.

The reason I reported the blocktim.c concern here is because I use Simulator (FreeRTOS\Demo\WIN32-MSVC\Win32.SLN) (and Visual Studio 2019) to sometimes debug my (STM32H753BIT6) code.
Also if one wants to port the STM32CubeMX / ST provided FreeRTOS code to any other microcontroller or simulator then there might become a concern, as the informed file may get used and the person porting the code will not be aware that configMAX_PRIORITIES will have to be updated.

Regards,
@rxa1031

closed 2020-09-23 by rxa1031