STMicroelectronics/STM32CubeH7 #60: STM32CubeMX IDE issue. Throws warning when using macros

rxa1031, 2020-09-24:
Hello,

Microcontroller used STM32H753BI

Issue: STM32CubeMX 6.0.0 Macro concerns

Case Number informed by ST (internal community): 00112804

I have still not heard back on this from ST Team and hence raising the concern here.

Regards,
@rxa1031

RKOUSTM, 2020-09-24:
Hi @rxa1031,

First, allow me to thank you for your contribution.

However, this issue is rather related to **CubeMX** than to the firmware published within this repository.
In fact, aspects related to CubeMX tool are not handled at GitHub level. To speed up the analysis procedure, this issue has been transferred to the CubeMx development teams for deeper analysis and to  obtain more details about the solution. We will be back to you as soon as we get an answer. 

With regards,

RKOUSTM, 2020-10-09:
Hi @rxa1031,

 Thank you for your contribution.  In order to allow a better understanding of the problem, could you please give us more details to allow the **CubeMX development teams** to reproduce this issue.

Thank you again for your contribution. 

With regards,

rxa1031, 2020-10-09:
Hello @RKOUSTM ,

Please refer to below link for more details:

https://community.st.com/s/question/0D53W00000JhsU4SAJ/stm32cubemx-ide-issue-throws-warning-when-using-macros

I do not want to make modifications to the IOC file just for changing / updating number of Standard and extended filters. Similarly I do not want to make changes to IOC file just for updating UART baudrate, and hence using macros.

In case the macro is an alias for a variable, the variable is declared globally in Main.C file and desired value is updated to the variable before the Init Function / Function using macro is called.

Regards,
Rajeev

rxa1031, 2020-10-09:
Hello @RKOUSTM 

The reason I reported this as an issue is because there are two different behaviors shown by STM32CubeMX.

The correct behavior is shown by STM32CubeMX when the IOC file is open and user first adds the macro (like dRS485PowerSupplyBaudRate) and then assigns the macro to one of the property (i.e. to Baud Rate). If the IOC file is saved and code is generated without closing the project, then STM32CubeMX does not throw any error.

The same STM32CubeMX version shows a different and incorrect behavior when the project is closed and reopened. Now:

if one has an already defined macro, which one wants to assign to a property, the STM32CubeMX will not let you assign the macro name to the property. (This is different from behavior stated earlier).
If one generates code, the STM32CubeMX throws warning for Devices which have macro names assigned to properties. (Again this behavior is different from earlier stated behavior).
 
My concern is that when the STM32CubeMX allows me to add Macros and then assign them to a property, why can it not leave them as is and:

Allow me use / assign the existing macro names even after I reopen project.
Inform in the warning about which macro it is considering to be a concern, so that I may decide on whether to disregard the warning or abort code generation.
 
Regards,
Rajeev

RKOUSTM, 2021-02-09:
Hi @rxa1031, 

Thank you for this report. This issue has been reported to the CubeMX team who will take in charge your request in the dedicated [section](https://community.st.com/s/question/0D53W00000JhsU4SAJ/stm32cubemx-ide-issue-throws-warning-when-using-macros) of ST Community.

As this is a CubeMX-related issue and not directly related to the firmware published in this repository, please allow me to close it. Thank you for your comprehension and contribution.

With regards,

closed 2021-02-09 by RKOUSTM