STMicroelectronics/STM32CubeH7 #61: Max value for configMAX_PRIORITIES

rxa1031, 2020-09-24:
Hi,

As per https://www.freertos.org/RTOS-task-priority.html:

> Each task is assigned a priority from 0 to ( configMAX_PRIORITIES – 1 )

It is informed under #59 that the reason to choose value 56 for configMAX_PRIORITIES is because the value corresponds to osPriorityISR.

However the above statement violates the FreeRTOS guideline stated above. The value of configMAX_PRIORITIES  shoule be:

`configMAX_PRIORITIES = osPriorityISR + 1`

Please suggest.

Thanks,
@rxa1031

RKOUSTM, 2020-09-24:
Hi @rxa1031,

Thank you for your contribution. Please note, that ``FreeRTOSConfig_template.h`` configuration file contain:
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Middlewares/Third_Party/FreeRTOS/Source/include/FreeRTOSConfig_template.h#L99-L107
Thus all user configurable priorities in ``CMSIS RTOS2`` can be used, from osPriorityLow to osPriorityRealtime7.
I hope this answers your question. Unfortunately we don't treat aspect related to `FreeRTOS in our GitHub repositories. As the Third_Party files are out of the scope of STMicroelectronics what about reporting this issue to the [FreeRTOS GitHub](https://github.com/freertos)?

With regards,

rxa1031, 2020-09-25:
Hello @RKOUSTM,

Does ST develope its own flavor of FreeRTOS which it bundles with STM32CubeMX. Where is the documentation for ST provided flavor of FreeRTOS?

I would like to have a copy of your document, so that I become aware of differences between your flavor of FreeRTOS and the FreeRTOS available on GitHub (Which also provided sample codes that execute on various STM32 series microcontrollers).

Regards,
@rxa1031

CCASTM, 2020-10-28:
Hello rxa1031,

ST proposes the usage of FreeRTOS in the context of the STM32Cube ecosystem. As mentioned in the st_readme.txt (under the Source directory, beside the original readme.txt)  ST insures only the adaptation to the cmsis_os layer and the validation of our examples projects, for any problem related to the OS itself we suggest to  contact FreeRTOS supplier as proposed by  my colleague RKOUSTM.
You can find more information on the porting and the usage in the context of STM32Cube in the Application Note UM1722
https://www.st.com/resource/en/user_manual/dm00105262-developing-applications-on-stm32cube-with-rtos-stmicroelectronics.pdf

Regards
Christophe

closed 2020-11-03 by RKOUSTM