STMicroelectronics/STM32CubeH7 #62: Requesting change to HAL_FDCAN_GetProtocolStatus() and FDCAN_COM_XXX macros

rxa1031, 2020-09-30:
Hello,

Please review the function `HAL_FDCAN_GetProtocolStatus()` implementation and the values assigned to macros `FDCAN_COM_STATE_SYNC`, `FDCAN_COM_STATE_IDLE`, `FDCAN_COM_STATE_RX`, and `FDCAN_COM_STATE_TX` (henceforth referred to as FDCAN_COOM_XXX).

Some observations:
1. Within function `HAL_FDCAN_GetProtocolStatus()` body, only the assignment made to `ProtocolStatus->Activity` is not `right shifted`. I understand this implementation is correct based on current values for `FDCAN_COM_XXX`. However, the values do not coincide with the values in datasheet, and without an API document containing this information, coder / user using the API may assume Activity value to be incorrect.

Requests:
1. Could the values assigned to FDCAN_COM_XXX macros be changed so that the valid values are 0 through 3:
```
#define FDCAN_COM_STATE_SYNC ((uint32_t)0x00000000U) /*!< Node is synchronizing on CAN communication */
#define FDCAN_COM_STATE_IDLE ((uint32_t)0x00000001U) /*!< Node is neither receiver nor transmitter   */
#define FDCAN_COM_STATE_RX   ((uint32_t)0x00000002U) /*!< Node is operating as receiver              */
#define FDCAN_COM_STATE_TX   ((uint32_t)0x00000003U) /*!< Node is operating as transmitter           */
``

2. May you please change `ProtocolStatus->Activity = (StatusReg & FDCAN_PSR_ACT);` to `ProtocolStatus->Activity = (StatusReg & FDCAN_PSR_ACT) >> FDCAN_PSR_ACT_Pos;`
for ease of understanding.

Please share your views.

Thanks,
@rxa1031

RKOUSTM, 2020-10-05:
Hi @rxa1031,

Thank you for your contribution.  Indeed, for ``stm32h7xx_hal_fdcan.c`` and  ``stm32h7xx_hal_fdcan.h`` driver code, there is no problem. As noted in the reference manual, the ``Activity`` bites that are used to monitors the module controller area network communication state are defined in the ``FDCAN protocol status register (FDCAN_PSR)`` as follows: 
[image]

So, we don't need to shift these bits in the ``stm32h7xx_hal_fdcan.c`` file and the writing to the LSE[2:0] bits has no effect on the bit value. Since the values assigned to ``FDCAN_COM_XXX`` macros in the ``stm32h7xx_hal_fdcan.h`` file are already correctly set based on the following configuration the Bits 4:3 **ACT[1:0]**:
- 00: Synchronizing: node is synchronizing on CAN communication
- 01: Idle: node is neither receiver nor transmitter
- 10: Receiver: node is operating as receiver
- 11: Transmitter: node is operating as transmitter 

Please allow me then to close this thread. Thank you for your comprehension and thank you again for your contribution.

With regards,

closed 2020-10-05 by RKOUSTM

rxa1031, 2020-10-05:
Hello @RKOUSTM 

I am looking at `structure FDCAN_ProtocolStatusTypeDef` and analyzing which macros to use for anayzing variable values for variables like `ProtocolStatus->BusOff`, `ProtocolStatus->ErrorPassive`, `ProtocolStatus->TDCvalue` etc.

ST has provided @ref / references only for few variables like LastErrorCode, DataLastErrorCode, and Activity.

As per ST shared explanation for FDCAM_COM_XXX, one would try to look for different macros declared for LEC[2:0] and DLEC[2:0] with different values (based on bit position) even for same error, (which I do not observe.).

In my earlier shared concern, I am discussing in terms of _code readability and coder understanding_. I am NOT reporting a bug in existing code, but I expect a change for ease of reading and understanding for all coders.

I am also informing that ST is following two different styles for declaring macros and using them. _I am only requesting ST to use one style out of the two._

I am unable to understand why ST want to have a different style of "declaring macro and assignment sequence for ACT[1:0] bits position [4:3]" and another style for declaring macros and assignment for Data Last Error Code[2:0] bits position [10:8].

I believe only one of the below two approaches should be used:
1. Either do not right shift any value that is assigned to any variable present with `structure FDCAN_ProtocolStatusTypeDef (variable name ProtocolStatus)` and use the macros that declare values corresponding to the bits position in use. (example ProtocolStatus->Activity and current value assigned to FDCAN_COM_STATE_TX)
2. Or right shift the value before assigning to any variable present `structure FDCAN_ProtocolStatusTypeDef (variable name ProtocolStatus)` and use the macros that define the values (and not place values) for the bits (example ProtocolStatus->DataLastErrorCode and current value assigned to FDCAN_PROTOCOL_ERROR_NO_CHANGE)

Please refer to the below macros and code pieces and decide on its readability and understanding for a coder, assuming he / she does not refer to comments provided for the variables.

```
/** @defgroup FDCAN_protocol_error_code FDCAN protocol error code
  * @{
  */
#define FDCAN_PROTOCOL_ERROR_NONE      ((uint32_t)0x00000000U) /*!< No error occurred         */
#define FDCAN_PROTOCOL_ERROR_STUFF     ((uint32_t)0x00000001U) /*!< Stuff error               */
#define FDCAN_PROTOCOL_ERROR_FORM      ((uint32_t)0x00000002U) /*!< Form error                */
#define FDCAN_PROTOCOL_ERROR_ACK       ((uint32_t)0x00000003U) /*!< Acknowledge error         */
#define FDCAN_PROTOCOL_ERROR_BIT1      ((uint32_t)0x00000004U) /*!< Bit 1 (recessive) error   */
#define FDCAN_PROTOCOL_ERROR_BIT0      ((uint32_t)0x00000005U) /*!< Bit 0 (dominant) error    */
#define FDCAN_PROTOCOL_ERROR_CRC       ((uint32_t)0x00000006U) /*!< CRC check sum error       */
#define FDCAN_PROTOCOL_ERROR_NO_CHANGE ((uint32_t)0x00000007U) /*!< No change since last read */
/**
  * @}
  */

/** @defgroup FDCAN_communication_state FDCAN communication state
  * @{
  */
#define FDCAN_COM_STATE_SYNC ((uint32_t)0x00000000U) /*!< Node is synchronizing on CAN communication */
#define FDCAN_COM_STATE_IDLE ((uint32_t)0x00000008U) /*!< Node is neither receiver nor transmitter   */
#define FDCAN_COM_STATE_RX   ((uint32_t)0x00000010U) /*!< Node is operating as receiver              */
#define FDCAN_COM_STATE_TX   ((uint32_t)0x00000018U) /*!< Node is operating as transmitter           */
/**
  * @}
  */
```

```
  ProtocolStatus->LastErrorCode = (StatusReg & FDCAN_PSR_LEC);
  ProtocolStatus->DataLastErrorCode = ((StatusReg & FDCAN_PSR_DLEC) >> FDCAN_PSR_DLEC_Pos);
  ProtocolStatus->Activity = (StatusReg & FDCAN_PSR_ACT);
...
  ProtocolStatus->ProtocolException = ((StatusReg & FDCAN_PSR_PXE) >> FDCAN_PSR_PXE_Pos);
  ProtocolStatus->TDCvalue = ((StatusReg & FDCAN_PSR_TDCV) >> FDCAN_PSR_TDCV_Pos);

```
Regards,
Rajeev