STMicroelectronics/STM32CubeH7 #63: REV_ID_Z missing in file stm32h7xx_hal.c

rxa1031, 2020-09-30:
Hello,

The User reference manual of STM32H753 informs value `DBGMCU->IDCODE == 0x1001` corresponds to `Revision Z` microcontroller. However the macro `REV_ID_Z` is missing from the header file.

Please add the macro.

Thanks,
@rxa1031

ASELSTM, 2020-10-06:
Hi @rxa1031,

Thank you for this report. It will be forwarded to our development teams for analysis. We will be back to you as soon as we get feedback.

Thank you for your patience and thank you again for your contribution.

With regards,

ASELSTM, 2020-10-06:
Hi @rxa1031,

These defines are used to customize the HAL behavior according to the current revision. In fact, Rev.Z behavior is same as Rev.Y this why REV_ID_Z has not been added in the  stm32h7xx_hal.h file.

More over, only **Revision Y** and **Revision V** are manufactured, other revisions are either internal or intermediate revisions. 

Please allow me then to close this thread. Thank you for your comprehension and thank you again for your contribution.

closed 2020-10-06 by ASELSTM