STMicroelectronics/STM32CubeH7 #64: Incorrect value assigned to __STM32H7xx_CMSIS_DEVICE_VERSION and adding __STM32H7xx_HAL_VERSION to a header file

rxa1031, 2020-09-30:
Hello,

Please refer to below lines of code:
https://github.com/STMicroelectronics/STM32CubeH7/blob/beced99ac090fece04d1e0eb6648b8075e156c6c/Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h7xx.h#L110-L113

Please change it to:
```
#define __STM32H7xx_CMSIS_DEVICE_VERSION        ((__STM32H7xx_CMSIS_DEVICE_VERSION_MAIN     << 24)\
                                      |(__STM32H7xx_CMSIS_DEVICE_VERSION_SUB1 << 16)\
                                      |(__STM32H7xx_CMSIS_DEVICE_VERSION_SUB2 << 8 )\
                                      |(__STM32H7xx_CMSIS_DEVICE_VERSION_RC))
```

Also if possible place add `__STM32H7xx_HAL_VERSION` to one of the header files so that the value can be used during conditional compile of C code.

Regards,
@rxa1031

RKOUSTM, 2020-10-02:
Hi @rxa1031,

Thank you for this report. The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

RKOUSTM, 2020-10-02:
ST Internal Reference: 93826

RKOUSTM, 2021-03-11:
Hi @rxa1031,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.9.0 of the STM32CubeH7 published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-11 by RKOUSTM