STMicroelectronics/STM32CubeH7 #65: STM32CubeMX 6.0.1 with STM32Cube MCU Package for STM32H7 Series version 1.8.0 adds bad value for Keil UVPROJX file

rxa1031, 2020-10-01:
Hello,

I am currently using STM32CubeMX version 6.0.1 with STM32Cube MCU Package for STM32H7 Series version 1.8.0. The generated code executes when using ULINKpro, but does not execute when Reset signal is given to microcontroller or Power to Microcontroller is recycled.

When I open a IOC project that was created with below STM32CubeMX version with STM32CubeMX latest version 6.0.1, it asks for migrating the project.
```
MxCube.Version=5.6.1
MxDb.Version=DB.5.0.60
```
Once I migrate the project I save and close it. Then open the IOC again and generate code.

Below **issue** is observed when Toolchain / IDE: **MDK-ARM** (minimum version 5.27) is selected and Source code is generated.
```
						<uC99>1</uC99>
						<uGnu>1</uGnu>
```

When the earlier STM32CubeMX (version 5.6.1 was in use along with STM32Cube MCU Package for STM32H7 Series version 1.7.0) the generated Keil UVPROJX file had below values:
```
						<uC99>1</uC99>
						<uGnu>0</uGnu>
```

Please fix the bug by ensuring that <uGnu>**0**</uGnu> is added to Keil UVPROJX file.

Regards,
@rxa1031

RKOUSTM, 2020-10-01:
Hi @rxa1031,

Glad to read from you again. Thank you for your contribution. However, this issue is related to a generation problem with Cube MX tool. In fact, aspects related to Cube MX tool are not handled at GitHub level. They are rather treated at **ST Community** level, precisely within the Cube [MX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated section.

Thank you for your comprehension.

With regards,

rxa1031, 2020-10-01:
Hello @RKOUSTM ,

Thanks for the information. I will share my concern on St Community.

Thanks,
@rxa1031

rxa1031, 2020-10-01:
Hello @RKOUSTM 

I am not sure whether closing this issue was a right step from my side.

Is there a way that this issue gets internally reported to appropriate team?

You may take action on this issue.

Regards,
@rxa1031

RKOUSTM, 2020-10-01:
Hi @rxa1031,

Thank you for your contribution. Allow me to close this ticket as it is out of the scope of this GitHub organization.

With regards,

closed 2020-10-01 by RKOUSTM