STMicroelectronics/STM32CubeH7 #67: STM32CubeH7 MDK-ARM Code behaves abnormally when MicroLIB is unselected

rxa1031, 2020-10-01:
Hello,

STM32CubeH7 Keil Code behaves abnormally when MicroLIB is unselected. Currently using STM32Cube MCU Package for STM32H7 Series version 1.8.0.

Toolchain IDE: MDK-ARM (minimum version 5.27).

I have tried confirming the behavior both with Keil Compiler Version 5 and Keil Compiler Version 6.

This abnormal behavior is NOT OBSERVED when ULINK Pro debugger is used for debugging code. Hence, unable to find the reason for the issue.

Please help.

Regards,
Rajeev

RKOUSTM, 2020-10-07:
Hi @rxa1031,

Thank you for your contribution. In order to allow a better analysis of this problem, could you please give us more details on how to reproduce it and could you please give us the name of the project that is used to reproduce this issue.  

Thank you again for your contribution.
 
With regards,

rxa1031, 2020-10-07:
Hello @RKOUSTM ,

I am sharing both the IOC files that I use.

[STM32CubeMX_Issues.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/5343423/STM32CubeMX_Issues.zip)

I am not in position o share the generated code along with propitiatory code. But these IOC files should help with:

1. Confirming that MicroLIB setting is not set when code is generated for MDK-ARM (minimum version 5.27) 
2. MicroLIB if already checked in UVPROJX / UVOPTX file (with say Keil Compiler Version 6 selected) then on regenerating the code with help of IOC file the MicroLIB check box get deselected. (Same is the case when Keil Compiler Version 5 is selected).
3. In generated file UVPROJX, both uC99 and uGnu tags are set to 1. I believe that uGnu should be set to 0.
4. STM32CubeMX does not allow generating code which can be compiled using Keil Compiler version 6, so one needs to open UVPROJX and UVOPTX files to change folder name RVDS to GCC and then copy the FreeRTOS GCC\ARM_CM4F to appropriate path based on the path name which has the folder GCC name mentioned. The FreeRTOS code then compiles without issues with Keil Compiler Version 6. However on regenerating the code with help of IOC file he GCC folder is deleted and RVDS folder and its paths are added back to project.

You may call me on my mobile number or have a ZOOM call, in case you may want a live demo.

Thanks and regards,
@rxa1031

RKOUSTM, 2020-10-08:
Hi @rxa1031, 

Thank you for this report. However, this is rather a Cube MX related issue and I am afraid we do not address it here. Please submit it to the **ST Community**, in the dedicated [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx). There you shall find people who will address your request and whom you will hopefully find a solution with.

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to our ecosystem (namely the Cube MX tool), please allow me to close it. 

Thank you for your contribution.  We are looking forward to reading from you again.

With regards,

rxa1031, 2020-10-08:
Hi @RKOUSTM 

I just noticed that even when all STM32CubeMX introduced issues (like uGnu value) are fixed, the code does not compile with MicroLib unselected.

Am I correct in understanding that `compile time issues` and function `assert_param(IS_NVIC_PREEMPTION_PRIORITY(PreemptPriority));` call leading to `assert_failed()` function call will be addressed under this project issues?

Please share your views.

Thanks,
Rajeev

closed 2020-10-26 by RKOUSTM