STMicroelectronics/STM32CubeH7 #68: HAL_UART_STATE_TIMEOUT not used by driver. Also it is set to 0xA0U and not to 0xC0U (i.e. bits [7:6] = 11)

rxa1031, 2020-10-03:
Hello,

I am trying to implement UART Receive Timeout, and observe that searching for `HAL_UART_STATE_TIMEOUT` only shows one line of code which is the macro declaration.

I observe below when searching for `HAL_UART_StateTypeDef` which is used in function `HAL_UART_GetState()`:

```
* @brief HAL UART State definition
 * @note HAL UART State value is a combination of 2 different substates: gState and RxState (see @ref UART_State_Definition).
 *    - gState contains UART state information related to global Handle management
 *     and also information related to Tx operations.
 *     gState value coding follow below described bitmap :
 *     b7-b6 Error information
 *       00 : No Error
 *       01 : (Not Used)
 *       10 : Timeout
 *       11 : Error
```

Based on above information I expect value of `HAL_UART_STATE_TIMEOUT should be 0xC0U (i.e. bits [7:6] = 11)` and not 0xA0U.

Please help me understand, why HAL_UART_STATE_TIMEOUT is set to 0xA0U.

Initialization:
static void MX_USART1_UART_Init(void)
{

  /* USER CODE BEGIN USART1_Init 0 */

  /* USER CODE END USART1_Init 0 */

  /* USER CODE BEGIN USART1_Init 1 */

  /* USER CODE END USART1_Init 1 */
  huart1.Instance = USART1;
  huart1.Init.BaudRate = dUSART1AsynchronousBaudrate;
  huart1.Init.WordLength = UART_WORDLENGTH_8B;
  huart1.Init.StopBits = UART_STOPBITS_1;
  huart1.Init.Parity = UART_PARITY_NONE;
  huart1.Init.Mode = UART_MODE_TX_RX;
  huart1.Init.HwFlowCtl = UART_HWCONTROL_NONE;
  huart1.Init.OverSampling = UART_OVERSAMPLING_16;
  huart1.Init.OneBitSampling = UART_ONE_BIT_SAMPLE_ENABLE;
  huart1.Init.ClockPrescaler = UART_PRESCALER_DIV10;
  huart1.AdvancedInit.AdvFeatureInit = UART_ADVFEATURE_NO_INIT;
  if (HAL_UART_Init(&huart1) != HAL_OK)
  {
    Error_Handler();
  }
  if (HAL_UARTEx_SetTxFifoThreshold(&huart1, UART_TXFIFO_THRESHOLD_7_8) != HAL_OK)
  {
    Error_Handler();
  }
  if (HAL_UARTEx_SetRxFifoThreshold(&huart1, UART_RXFIFO_THRESHOLD_1_8) != HAL_OK)
  {
    Error_Handler();
  }
  if (HAL_UARTEx_EnableFifoMode(&huart1) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN USART1_Init 2 */

		if(IS_UART_RECEIVER_TIMEOUT_VALUE(2)) // earlier value dUSART1ReceiveTimeoutInBitsCount))
		{
			HAL_UART_ReceiverTimeout_Config(&huart1, 2);// #DRA: modified for testing dUSART1ReceiveTimeoutInBitsCount);
			HAL_UART_EnableReceiverTimeout(&huart1);
			__HAL_UART_CLEAR_IT(&huart1, UART_CLEAR_RTOF);
			SET_BIT(huart1.Instance->CR1, USART_CR1_RTOIE);
		}
		else
		{
			dLEDShowError(dUSART1RxTimeoutTooLarge);
		}
	/* USER CODE END USART1_Init 2 */
}
```
```
		if(dRxComplete == bIsAsyncUSART1ReceiveComplete)
		{
			bIsAsyncUSART1ReceiveComplete = dRxInProgress;
		}
		else
		{
			dLEDShowError(dBadUSART1RxComStateBeforeRxInitiate);
			assert_failed((uint8_t *)__FILE__, __LINE__);
		}

		__schedule_barrier();

		if(HAL_OK != HAL_UART_Receive_IT(&huart1, (uint8_t *)ucTestKey, 1))
		{
			dLEDShowError(dUSART1RxInitiationFailed);
			assert_failed((uint8_t *)__FILE__, __LINE__);
		}

		__schedule_barrier();
//
// Here I want to check for HAL_UART_STATE_TIMEOUT, but function never returns HAL_UART_STATE_TIMEOUT.
//
		while (HAL_UART_STATE_READY != HAL_UART_GetState(&huart1))
		{
			__NOP();
		}

		__schedule_barrier();
		while(dRxInProgress == bIsAsyncUSART1ReceiveComplete)
		{
			__NOP();
		}
```

Please suggest a fix or a method for enabling Asynchronous UART Receive (interrupt) timeout.

Regards,
Rajeev

ASELSTM, 2020-10-05:
Hi @rxa1031,

Actually, the point you raised has already been raised in another issue #31. We have discussed this internally and we are working on the implementation of a service to enable Reception Timeout interrupt. A fix will be made available in the frame of a future release.

Regarding the second point, indeed, an error has been introduced in the comment which should rather be :
```
 *    b5     Peripheral initialization status
 *        0  : Reset (Peripheral not initialized)
```
```diff
-*        1  : Init done (Peripheral not initialized. HAL UART Init function already called)
+*        1  : Init done (Peripheral initialized. HAL UART Init function already called)
```
The same correction should be applied in the description of **RxState** bitmap.  Thus, `HAL_UART_STATE_TIMEOUT` should be set to **0xA0U** and not to  0x80U since b5=1. 

With regards,

ASELSTM, 2020-10-05:
ST Internal Reference: 86997

ASELSTM, 2020-10-05:
ST Internal Reference: 93918