STMicroelectronics/STM32CubeH7 #70: Compile time linker errors

rxa1031, 2020-10-08:
Hello,

Using Latest version of STM32CubeMx and STM32H7 package.

After enabling FreeRTOS and making minor changes to its settings within the STM32CubeMX, code is generated for MDK-ARM (minimum version 5.27).

On compiling the code below ERRORS are observed:
```
linking...
Gateway\Gateway.axf: Error: L6200E: Symbol vApplicationIdleHook multiply defined (by cmsis_os2.o and freertos.o).
Gateway\Gateway.axf: Error: L6200E: Symbol vApplicationMallocFailedHook multiply defined (by cmsis_os2.o and freertos.o).
Gateway\Gateway.axf: Error: L6200E: Symbol vApplicationStackOverflowHook multiply defined (by cmsis_os2.o and freertos.o).
Gateway\Gateway.axf: Error: L6200E: Symbol vApplicationTickHook multiply defined (by cmsis_os2.o and freertos.o).
Not enough information to list image symbols.
Not enough information to list load addresses in the image map.
Finished: 2 information, 0 warning and 4 error messages.
```

Attached IOC file for your reference.

[Gateway.zip](https://github.com/STMicroelectronics/STM32CubeH7/files/5345757/Gateway.zip)

Before compiling code add below to main.c under `/* USER CODE BEGIN PV */`

```
uint8_t u8MaxStandardCAN1Filters;
uint8_t u8MaxExtendedCAN1Filters;
uint8_t u8MaxStandardCAN2Filters;
uint8_t u8MaxExtendedCAN2Filters;
```

Please help fix the issues.

Thanks,
Rajeev

rxa1031, 2020-10-08:
Hello,

`Fix for: Gateway\Gateway.axf: Error: L6200E: Symbol vApplicationIdleHook multiply defined (by cmsis_os2.o and freertos.o).`

When `configUSE_IDLE_HOOK == 0` the function is only available in cmsis_os2.c and is not considered during compilation.

When `configUSE_IDLE_HOOK == 1` the linker error is triggering.

Please remove below from cmsis_os2.c the cmsis_os2.c file.

```
#if (configUSE_IDLE_HOOK == 1)
__WEAK void vApplicationIdleHook (void){}
#endif
```
The function body is available in _Src\freertos.c starting from line number 78_

rxa1031, 2020-10-08:
One more observation:

If both UART HAL Driver (say for USART1) and UART LL Driver (say for UART8, USART3, and USART6) are used then additional Linker errors are observed:

```
.\Gateway-Objects\Gateway.axf: Error: L6200E: Symbol SystemInit multiply defined (by system_stm32h7xx_1.o and system_stm32h7xx.o).
.\Gateway-Objects\Gateway.axf: Error: L6200E: Symbol SystemCoreClockUpdate multiply defined (by system_stm32h7xx_1.o and system_stm32h7xx.o).
.\Gateway-Objects\Gateway.axf: Error: L6200E: Symbol SystemD2Clock multiply defined (by system_stm32h7xx_1.o and system_stm32h7xx.o).
.\Gateway-Objects\Gateway.axf: Error: L6200E: Symbol D1CorePrescTable multiply defined (by system_stm32h7xx_1.o and system_stm32h7xx.o).
.\Gateway-Objects\Gateway.axf: Error: L6200E: Symbol SystemCoreClock multiply defined (by system_stm32h7xx_1.o and system_stm32h7xx.o).
```

I am using a third party Library which needs the use of UART / USART LL Driver and hence I cannot switch to UART / USART HAL Driver. Also for ease of communication with external device over UART, I want to continue using UART HAL Driver.

Please suggest how to fix this issue.

Regards,
Rajeev

rxa1031, 2020-10-08:
May someone help me understand reason for below:

Rebuild started: Project: Gateway
*** Using Compiler 'V6.14', folder: 'C:\Keil_v5\ARM\ARMCLANG\Bin'
Rebuild target 'Gateway'
**Note: source file '../Src/system_stm32h7xx.c' - object file renamed from '.\Gateway-Objects\system_stm32h7xx.o' to '.\Gateway-Objects\system_stm32h7xx_1.o'.**

ASELSTM, 2020-10-20:
Hi @rxa1031,

Thank you for this other report. It has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

Best regards,

rxa1031, 2021-03-23:
Hello @ALABSTM ,

Are there any findings / updates that you may be able to share with me.

Thanks,
Rajeev

ASELSTM, 2021-07-15:
Hi @rxa1031,

Thank you for this report. The issue you pointed out has been confirmed, a fix will be implemented and made available in future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2021-07-15:
ST Internal Reference: 110640