STMicroelectronics/STM32CubeH7 #71: Calling SystemClock_Config() results in call to assert_failed() because of missing assignment uwTickPrio = TICK_INT_PRIORITY

rxa1031, 2020-10-10:
Hello @RKOUSTM ,

After a generated code with help of STM32CubeMX (with STM32Cube MCU Package for STM32H7 Series version 1.8.0), which has FreeRTOS enabled and sues CMSIS V2, I _added a endless while loop_ to both the `assert_failed()` and the `Error_Handler()` functions.

When debugging the code it is observed that `assert_failed()` was triggering when `SystemClock_config()` executed.

It looks like the HAL library got updated to use the variable 'uwTickPrio' in the function HAL_RCC_ClockConfig, instead of what was used in the older library, `TICK_INT_PRIORITY`.

May ST please help me understand the reason why, 'uwTickPrio' is intentionally initialized to an invalid value.

In order to fix the (assert_failed() function called) issue I have added 
```
	uwTickPrio = TICK_INT_PRIORITY;
```
just before the comment:
```
  /* USER CODE END Init */
```

The code now works fine.

Please help me with your inputs.

Thanks,
Rajeev

RKOUSTM, 2020-11-02:
Hi @rxa1031,

Thank you for this report. We tried to reproduce the issue you described, but we couldn't notice the problem.

According to the following process of initializing the "uwTickPrio" variable to an invalid value, an update of the time base management is proposed to take into account low frequency systems where, at low frequency, a 1 ms systick interrupt does not allow to do anything else than a treated interrupt. 
```
uint32_t uwTickPrio   = (1UL << __NVIC_PRIO_BITS); /* Invalid PRIO */
```
In HAL_InitTick() function, when checking the tick priority, it is better to relay on __NVIC_PRIO_BITS to check the tick priority limit of ticks instead of the initialized number (3 for M0, 15 for M4/M7).

Thank you again for your contribution.

With regards,

rxa1031, 2020-11-02:
Hi @RKOUSTM ,

I believe the reason for not observing the issue is that, either the FULL ASSERT related macro was not enabled or the tester did not place a break point inside the assert_failed() function body (or both).

Please confirm whether, the tester enable FULL ASSERT in your test code? 

stm32h7xx_hal_conf.h must include:

 #define USE_FULL_ASSERT    1U

Did the tester place a break point or a debug output message inside assert_failed() function?

The declaration for the function is:

void assert_failed(uint8_t *file, uint32_t line)

Regards,
Rajeev

rxa1031, 2020-11-02:
Hi @RKOUSTM ,

I observe below in STM32CubeH7/Drivers/CMSIS/Device/ST/STM32H7xx/Include/stm32h753xx.h

`#define __NVIC_PRIO_BITS          4       /*!< CM7 uses 4 Bits for the Priority Levels       */`

Based on your input I assume the macro should have been assigned the value 15 instead of 4.

Please suggest.

Thanks,
Rajeev

rxa1031, 2020-11-02:
1<<3 is 8 and 1<<4 is 16. In your comment I observe values as 3 and 15 and hence not sure whether you are asking me to left shift 1 by 15 or by 4 (for M7).

RKOUSTM, 2020-11-03:
Hi @rxa1031, 

Thank you for this report, once again. Indeed, concerning  cm4 and cm7, **__NVIC_PRIO_BITS** is configured to 4. Regarding the point you reported, we couldn't reproduce the issue you described. In this case, it is rather related to **Cube MX** issue or even way you think to design you application. Unfortunately we don't treat aspect related to **CubeMx** tool in our GitHub repositories. Please allow me to redirect you to the **ST Community**. You can report this issue in the CubeMX related [section](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx). 

Please allow me to close this ticket as it is out of the scope of this GitHub organization. 
Thank you again for your report. We are looking forward to reading from you again.

With regards,

closed 2020-11-11 by RKOUSTM

rxa1031, 2020-11-26:
Hello @RKOUSTM ,

I found out that the reason for the primary / first concern is the issue mentioned under #23 

Regards,
@rxa1031

jnewcomb, 2021-01-02:
@rxa1031 
Hi.. I'm also experiencing this issue - and I fixed it by following your suggestion.
```
uwTickPrio = TICK_INT_PRIORITY;
```
Did you ever find out the root cause of this? I just followed the exact path you took!
Guess I just want to understand as it killed a whole day of my productivity - or should I just move on, none-the-wiser!
Thanks.
Jon.

rxa1031, 2021-01-04:
Hello Jon @jnewcomb ,

I am unable to figure out how to share the links to the concerns that I had raised regarding this issue on ST Community. Below is my profile page on ST Community.

https://community.st.com/s/profile/0050X000009yi2mQAA

**Concern 1:**
Is there a setting with STM32CubeMX to change the Tick Init Priority / TICK_INT_PRIORITY value?

Raised on November 27, 2020

Answer: No

**Concern 2:**
TickPriority not assigned to uwTickPrio

Contains two issues

_Raised on  November 19, 2020_

Requested modification of code present with Function **HAL_InitTick()**

Solution to the informed problem is to replace code:
```
    /*Configure the TIM2 IRQ priority */
      HAL_NVIC_SetPriority(TIM2_IRQn, TickPriority, 0);
     
      /* Enable the TIM2 global Interrupt */
      HAL_NVIC_EnableIRQ(TIM2_IRQn);
     
      /* Enable TIM2 clock */
      __HAL_RCC_TIM2_CLK_ENABLE();
``` 
with
```
    /*Configure the TIM2 IRQ priority */
      if (TickPriority < (1UL << __NVIC_PRIO_BITS))
      {
        HAL_NVIC_SetPriority(TIM2_IRQn, TickPriority ,0U);
     
        /* Enable the TIM2 global Interrupt */
        HAL_NVIC_EnableIRQ(TIM2_IRQn);

        uwTickPrio = TickPriority;
     
        /* Enable TIM2 clock */
        __HAL_RCC_TIM2_CLK_ENABLE();
      }
      else
      {
        return HAL_ERROR;
      }
```

We need the assignment `uwTickPrio = TickPriority;` added to function to fix the issue.

This issue is (possibly) fixed in STM32CubeMX for STM32H7XX Series Microcontrollers.

Regards,
Rajeev