STMicroelectronics/STM32CubeH7 #72: STM32H753BIT6 Temperature Sensor Channel different reported values

rxa1031, 2020-10-13:
Hello @RKOUSTM ,

I have enabled the ADC3 to read VBAT, VREF and Temperature Sensor Channel in continuous mode. After initialization of the ADC3, my code calls the calibration function once id starts reading the channels.

I am observing that `__HAL_ADC_CALC_TEMPERATURE()` outputs a lower temperature value compared to `__HAL_ADC_CALC_TEMPERATURE_TYP_PARAMS()`.

The value returned by `__HAL_ADC_CALC_TEMPERATURE()` seems to be closer to the temperature measured on the microcontroller's casing.

I assume the temperature of the _microcontroller DIE_ would be much higher. Please help me understand _which of the two formulas to use, and why is there a difference in the reported Temperature values_?

Below values will be helpful for analyzing the Temperature outputs:
```
VRef voltage is 3291 mV.
VBAT voltage is 712 mV.
40, and 45 are Temperature in Degree C calculated using two different formulas.
```

Test code currently in use follows:
```
main()
{
  /* Reset of all peripherals, Initializes the Flash interface and the Systick. */
  HAL_Init();

  /* USER CODE BEGIN Init */

  /* USER CODE END Init */

  /* Configure the system clock */
  SystemClock_Config();

  /* USER CODE BEGIN SysInit */

  /* USER CODE BEGIN 2 */

MX_ADC3_Init();
//--- user code continues ---

// --- ADC3 read Code ---
					uint16_t u16VRefMilliVoltsValue = 0;
					uint16_t u32VBATMilliVolts = 0;

					// Reading ADC3 inputs VBAT, VRef, and Internal Temperature Sensor Channel.

					if( HAL_OK != HAL_ADC_Start(&hadc3))
					{
						dLEDShowError(dADC3StartFailed);
						assert_failed((uint8_t *)__FILE__, __LINE__);
					}
					if(HAL_OK != HAL_ADC_PollForConversion(&hadc3, ADC_READ_FUNCTION_TIMEOUT))
					{
						dLEDShowError(dADC3PollForConversionFailed);
						assert_failed((uint8_t *)__FILE__, __LINE__);
					}
					g_u32ReadVRefDigitalValue = HAL_ADC_GetValue(&hadc3);

					if(HAL_OK != HAL_ADC_PollForConversion(&hadc3, ADC_READ_FUNCTION_TIMEOUT))
					{
						dLEDShowError(dADC3PollForConversionFailed);
						assert_failed((uint8_t *)__FILE__, __LINE__);
					}
					g_u32ReadVBATDigitalValue = HAL_ADC_GetValue(&hadc3);

					// Measure Temperature
					if(HAL_OK != HAL_ADC_PollForConversion(&hadc3, ADC_READ_FUNCTION_TIMEOUT))
					{
						dLEDShowError(dADC3PollForConversionFailed);
						assert_failed((uint8_t *)__FILE__, __LINE__);
					}
					g_u32ReadTemperatureDigitalValue = HAL_ADC_GetValue(&hadc3);

					HAL_ADC_Stop(&hadc3);

					// Calculating VRef voltage
					u16VRefMilliVoltsValue = __HAL_ADC_CALC_VREFANALOG_VOLTAGE(g_u32ReadVRefDigitalValue, ADC_RESOLUTION_16B);
					dOutputFormattedHWTestMessage("\nVRef voltage is %u mV.", u16VRefMilliVoltsValue);
					// Calculate battery voltage
					u32VBATMilliVolts = 4UL * __HAL_ADC_CALC_DATA_TO_VOLTAGE(u16VRefMilliVoltsValue, g_u32ReadVBATDigitalValue, ADC_RESOLUTION_16B);
					dOutputFormattedHWTestMessage("\nVBAT voltage is %u mV.", u32VBATMilliVolts);
					// Calculate battery voltage
					dOutputFormattedHWTestMessage("\r\n%ld, and %d are Temperature in Degree C calculated using two different formulas."
							, __HAL_ADC_CALC_TEMPERATURE(u16VRefMilliVoltsValue, g_u32ReadTemperatureDigitalValue, ADC_RESOLUTION_16B)
							, __HAL_ADC_CALC_TEMPERATURE_TYP_PARAMS(2000, 620, 30, u16VRefMilliVoltsValue, g_u32ReadTemperatureDigitalValue, ADC_RESOLUTION_16B));

  /* USER CODE END 2 */

}
```

```
/**
  * @brief ADC3 Initialization Function
  * @param None
  * @retval None
  */
static void MX_ADC3_Init(void)
{

  /* USER CODE BEGIN ADC3_Init 0 */

  /* USER CODE END ADC3_Init 0 */

  ADC_ChannelConfTypeDef sConfig = {0};

  /* USER CODE BEGIN ADC3_Init 1 */

  /* USER CODE END ADC3_Init 1 */
  /** Common config
  */
  hadc3.Instance = ADC3;
  hadc3.Init.ClockPrescaler = ADC_CLOCK_ASYNC_DIV1;
  hadc3.Init.Resolution = ADC_RESOLUTION_14B;
  hadc3.Init.ScanConvMode = ADC_SCAN_ENABLE;
  hadc3.Init.EOCSelection = ADC_EOC_SINGLE_CONV;
  hadc3.Init.LowPowerAutoWait = DISABLE;
  hadc3.Init.ContinuousConvMode = ENABLE;
  hadc3.Init.NbrOfConversion = 3;
  hadc3.Init.DiscontinuousConvMode = DISABLE;
  hadc3.Init.ExternalTrigConv = ADC_SOFTWARE_START;
  hadc3.Init.ExternalTrigConvEdge = ADC_EXTERNALTRIGCONVEDGE_NONE;
  hadc3.Init.ConversionDataManagement = ADC_CONVERSIONDATA_DR;
  hadc3.Init.Overrun = ADC_OVR_DATA_PRESERVED;
  hadc3.Init.LeftBitShift = ADC_LEFTBITSHIFT_2;
  hadc3.Init.OversamplingMode = ENABLE;
  hadc3.Init.Oversampling.Ratio = 8;
  hadc3.Init.Oversampling.RightBitShift = ADC_RIGHTBITSHIFT_3;
  hadc3.Init.Oversampling.TriggeredMode = ADC_TRIGGEREDMODE_SINGLE_TRIGGER;
  hadc3.Init.Oversampling.OversamplingStopReset = ADC_REGOVERSAMPLING_CONTINUED_MODE;
  if (HAL_ADC_Init(&hadc3) != HAL_OK)
  {
    Error_Handler();
  }
  /** Configure Regular Channel
  */
  sConfig.Channel = ADC_CHANNEL_VREFINT;
  sConfig.Rank = ADC_REGULAR_RANK_1;
  sConfig.SamplingTime = ADC_SAMPLETIME_2CYCLES_5;
  sConfig.SingleDiff = ADC_SINGLE_ENDED;
  sConfig.OffsetNumber = ADC_OFFSET_NONE;
  sConfig.Offset = 0;
  sConfig.OffsetSignedSaturation = DISABLE;
  if (HAL_ADC_ConfigChannel(&hadc3, &sConfig) != HAL_OK)
  {
    Error_Handler();
  }
  /** Configure Regular Channel
  */
  sConfig.Channel = ADC_CHANNEL_VBAT;
  sConfig.Rank = ADC_REGULAR_RANK_2;
  sConfig.SamplingTime = ADC_SAMPLETIME_387CYCLES_5;
  if (HAL_ADC_ConfigChannel(&hadc3, &sConfig) != HAL_OK)
  {
    Error_Handler();
  }
  /** Configure Regular Channel
  */
  sConfig.Channel = ADC_CHANNEL_TEMPSENSOR;
  sConfig.Rank = ADC_REGULAR_RANK_3;
  if (HAL_ADC_ConfigChannel(&hadc3, &sConfig) != HAL_OK)
  {
    Error_Handler();
  }
  /* USER CODE BEGIN ADC3_Init 2 */

		// #DRA: NOTE: ADC_CALIB_OFFSET_LINEARITY, i.e. ADCALLIN = 1 executes both the linearity and offset calibration.
		// ADC_SINGLE_ENDED, i.e. ADCALDIF launches a calibration which is only applicable for single-ended input conversion.
		if(HAL_OK != HAL_ADCEx_Calibration_Start(&hadc3, ADC_CALIB_OFFSET_LINEARITY, ADC_SINGLE_ENDED))
		{
			dLEDShowError(dADC3CalibrationFailed);
			assert_failed((uint8_t *)__FILE__, __LINE__);
		}

	#endif // of defined(_DRA_VBAT_VREF_AND_TEMPERATURE_ADC_READ_)
}
```

Please help with resolving temperature output valueconcerns.

Thanks,
Rajeev

rxa1031, 2020-10-13:
As the Battery had got discharged I have removed the battery. I am observing the VBAT reading around 712 mV and not Zero.

I am not sure why this is the behavior.

The VSS and VSSA ground are same. The battery negative is connected to VSS and battery +ve is connected to Anode of IN5819. The Cathode of 1N5819 is then connected to microcontroller's VBAT pin (i.e. Pin 6).

ASELSTM, 2020-11-13:
Hi @rxa1031,

Thank you for this report. Your request has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

With regards,

ASELSTM, 2020-11-16:
Hi @rxa1031,

For a further understanding of the problem, would you please check whether the chip is calibrated or not.  
For that, you have to access to the memory addresses where **TS_CAL1** and **TS_CAL2** values are stored and check if these values are not equal to zero and share them with us. 
You may find these memory addresses in the [datasheet](https://www.st.com/resource/en/datasheet/stm32h753vi.pdf) under the **Temperature sensor characteristics** section.  

[image]

With regards,

rxa1031, 2020-11-17:
Hello @ASELSTM  ,

Below are the observed values:

| Address | Value |
| 0x1FF1E820 | 0x78 |
| 0x1FF1E821 | 0x30 |
| 0x1FF1E840 | 0x7A |
| 0x1FF1E841 | 0x3F |

Regards,

ASELSTM, 2020-12-15:
Hi @rxa1031,

Thank for these details. Unfortunately we couldn't reproduce this issue. We used the same code and configuration that you shared and the temperature values returned by both macros are quite similar. The difference is minor, only 1°C. 

However, to answer your question regarding the difference between both macros : 

- **__HAL_ADC_CALC_TEMPERATURE** : The temperature computation within this macro is using temperature sensor calibration values stored in system memory for each device during production. These calibration values are the temperature sensor raw data acquired value at certain temperature calibrated in factory (30°C and 110°C for STM32H7 case) . The formula used for the temperature computation is detailed in the section **"25.4.33 Temperature sensor"** of the [reference manual](https://www.st.com/resource/en/reference_manual/dm00314099-stm32h742-stm32h743-753-and-stm32h750-value-line-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf). [image]
The calibration is not performed on some device during production, which is not your case since there is calibration values stored in the dedicated memory address. 

- **__HAL_ADC_CALC_TEMPERATURE_TYP_PARAMS** : The temperature computation within this macro is based on the temperature sensor typical values detailed in the [datasheet ](https://www.st.com/resource/en/datasheet/stm32h753vi.pdf) within the section **"6.3.23 Temperature sensor characteristics"** and the formula is described at the beginning of the **__LL_ADC_CALC_TEMPERATURE_TYP_PARAMS** macro description.

These two macros give quite similar temperature values, and the difference is minimal. 

I hope this will help you better understand the difference between these 2 macros. 

With regards,

ASELSTM, 2021-01-12:
Hi @rxa1031,

Please allow me to close this thread as no activity has been registered since a month. You can reopen it at any time if you have details to provide us to help you solve this issue.
Thank you for your comprehension and thank you once more for your contribution

With regards,

closed 2021-01-12 by ASELSTM